Atmosphere-NX/Atmosphere #696: panic occured while running atmosphere

LunaerD, 2019-11-11:
## Bug Report

N/A

### What's the issue you encountered?

 [Today I downloaded an app from the hb app store and then atmosphere crashd and when I try to launch emummc atmosphere says "panic occured while running atmosphere".</br>
I tried to delete the .nro app with my pc but I can't. </br>

### How can the issue be reproduced?

 * </br>
 1-Go to the hb app store
   2-download the BrowseNx app
   3- at the end of the download it crash. </br>

### Crash Report

</br>
https://gist.github.com/LunaerD/a1a4ba377d43536f1ca79fb3061c7e59</br>
</br>

### System Firmware Version

0.9.4</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: official release
  -  Official release version 5.0.2
- Do you have additional kips or sysmodules you're loading: nope
- Homebrew software installed:   no

### Additional context?

- Additional info about your environment: a friend installed the crack for me and it worked perfectly before the app download

SciresM, 2019-11-13:
Seems like a duplicate of #695?

closed 2019-11-13 by SciresM

LunaerD, 2019-11-13:
> ## Bug Report
> N/A
> 
> ### What's the issue you encountered?
> [Today I downloaded an app from the hb app store and then atmosphere crashd and when I try to launch emummc atmosphere says "panic occured while running atmosphere".
> 
> I tried to delete the .nro app with my pc but I can't.
> 
> ### How can the issue be reproduced?
> * 
> 
> 1-Go to the hb app store
> 2-download the BrowseNx app
> 3- at the end of the download it crash.
> 
> ### Crash Report
> 
> https://gist.github.com/LunaerD/a1a4ba377d43536f1ca79fb3061c7e59
> 
> ### System Firmware Version
> 0.9.4
> 
> ### Environment?
> * What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
> * Official release or unofficial build: official release
>   
>   * Official release version 5.0.2
> * Do you have additional kips or sysmodules you're loading: nope
> * Homebrew software installed:   no
> 
> ### Additional context?
> * Additional info about your environment: a friend installed the crack for me and it worked perfectly before the app download

> Seems like a duplicate of #695?

Nope because my crash is during the loading of atmosphere after I send a payload into my switch so I can't even load the crack

SciresM, 2019-11-13:
"Load the crack" - what?

Are you sure you're using a completely stock latest atmosphere release downloaded from ams repo? What version of atmosphere are you using?

Your crash report is an assert in the bsdsockets sysmodule, which suggests some homebrew is doing something weird.

What folders do you have in /atmosphere/titles?

LunaerD, 2019-11-13:
> "Load the crack" - what?
>
> Are you sure you're using a completely stock latest atmosphere release downloaded from ams repo? What version of atmosphere are you using?
>
> Your crash report is an assert [...]

I'm using the latest atmosphere releases for the 9.0.1 switch, and all of the folders are the same as when I installed the whole thing except the .nro app that I cannot delete.

ZachyCatGames, 2019-11-13:
Delete /atmosphere/titles/00FF747765616BFF. BrowseNX comes with a sysmodule that causes a fatal error on startup on 9.0+.

Grayfoxst, 2020-01-15:
Hello , im having the same issue , however when i browse the /Atmosphere/titles/ the folder is empty , and im not seeing any hidden file. any ideas?

dennthecafebabe, 2020-01-15:
`atmosphere/titles` is called `atmosphere/contents` now

Grayfoxst, 2020-01-15:
Thanks i already resolve it by doing it !!!

Bastiongg, 2020-01-16:
@ZachyCatGames Thank you this worked for me 💯

DSMan97, 2020-01-30:
> > Delete /atmosphere/titles/00FF747765616BFF, that’ll solve the error your getting on boot. BrowseNX comes with a sysmodule that causes a fatal error on startup on 9.0+.
> > @ZachyCatGames Thank you this worked for me 💯

So 00FF747765616BFF correspond to BrowseNX ?? Jus to know for the next timw, thanks in advance :)

msamkov, 2020-05-20:
Помогло удаление каталога /atmosphere/contents/00FF747765616BFF