Atmosphere-NX/Atmosphere #698: Date/Time in fatal reports.

fennectech, 2019-11-15:
## Feature Request

### What feature are you suggesting?
#### Overview:
Date and time in fatal reports

#### Nature of Request:  
- Addition
Add date and time to fatal reports in an easily readable fashion

### What component do you feel this would best fit within?
-(https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>
### Why would this feature be useful?
I've done a hell of a lot of testing and debugging and i would find time/date useful  Having it in the attributes can result in the data getting stripped in some cases   and is entirely lost when transferring over the internet

shchmue, 2019-11-15:
inb4 you can't pull time info in that context

SciresM, 2019-11-15:
It's encoded in the filename, man.

closed 2019-11-15 by SciresM

fennectech, 2019-11-15:
Mk. im stupid. Ill have to look at it harder xD