Atmosphere-NX/Atmosphere #700: Feature Request: "MTP"/SD card management over USB

NWPlayer123, 2019-11-21:
## Feature Request

### What feature are you suggesting?
#### Overview:
A service(?) that lets you plug your Switch into a computer and have it mount the SD card to copy, delete, move, etc. like if the actual SD was inserted, ala "Media Transfer Protocol" on phones.</br>

#### Smaller Details: 
As mentioned, it should probably be a service that's always running, looks like there's https://github.com/Atmosphere-NX/Atmosphere/pull/643 but it hasn't been touched in a while and the [new-ipc](https://github.com/Atmosphere-NX/Atmosphere/tree/new-ipc) tree will probably mess that up. </br>
I'm not sure how much fuckery would need to go into e.g. Windows, Mac to get them to work correctly, but theoretically should work out of the box. </br>

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>

### Why would this feature be useful?
It would make things a lot easier than e.g. FTP, which needs an actual program, or a "third-party" homebrew thing.</br>

SciresM, 2019-11-21:
#643 is actually unrelated to MTP stuff.

Anyway, providing access to SD card via USB is eventually planned for ams.tma.

That's still blocked on cross-platform USB.

SciresM, 2019-11-21:
Tentatively gonna close this, pending the usual ams.TMA discussions. Ping me on discord whenever the cross-platform USB stuff gets solved.

closed 2019-11-21 by SciresM