Atmosphere-NX/Atmosphere #701: Battery Module

ShadowOne333, 2019-11-22:
## Feature Request

### What feature are you suggesting?
#### Overview:
- Some sort of module / code that makes Horizon OS to read the proper capacity and voltages values from the battery and, given the values obtained directly from the (i2c?) capacity/voltage of the battery, force Horizon OS to display the proper battery percentage into the OS itself.
This includes the capacity full and capacity design values.

Ex.: In my case, since a few weeks ago, my Switch started giving wrong values for capacity design/full, making the percentage inside Horizon OS to be wrong at all times, making the Switch to turn off itself at random values of percentage (40%, 30%). For instance, my capacity design and full values tend to jump around to different values (8k, 9k, 10k, 13k, etc), making the percentage display useless to determine the exact battery remaining on the console.

However, the battery itself does work as intended. It plays for the standard amount of hours in certain titles (3-4 hours in heavy games), and the "voltage/capacity now" values read by Hekate from the console are within the normal range (3200mV - 4200mV, with Horizon OS dying from low battery at around 3600mV as expected).

I have tried running several tests made by CTCaer himself of Hekate to reset the battery readings, and while they do show up properly the first time I run the reset, after a normal start and reboot, the capacity full/design go back to values above the normally expected ones.
I have also opened up the Switch and disconnected and reconnected the battery to try to reset the readings as well, with the same result happening (design/full being above normal).

I'm not sure if HOS is storing said values somewhere in the Switch, but it could be the case.

#### Smaller Details: 
- Hekate on its own has code to read the values of capacity and voltage of the battery from the Switch in detail. Perhaps this could be used as a starting point for said implementation.

#### Nature of Request:  
- Addition
  - Implementation of a module or code that forces a proper capacity design, capacity full values of Horizon OS. 
  - Also, as a second option, make the percentage display within HOS to read the values of voltage properly to display the adequate battery remaining in the console.

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>
- [**Troposphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Application-level patches to the Horizon OS.</br>

### Why would this feature be useful?
This feature would help people that for some reason got their capacity full / capacity design out of sync from the OS (or are getting problems with battery values being out of sync with the values read by Horizon OS), so that the proper battery percentages  are read and applied properly into the OS based on the true capacity and voltage values of the battery.

Resaec, 2019-12-05:
If you really screwed up your battery calibration, the only thing to correct that is to unplug the battery for a few minutes.
You can check that in Hekate. If it shows a very low "full capacity" you should try that.

ShadowOne333, 2019-12-06:
I did that.
Unplugged the battery manually by opening up the Switch, disconnecting it from the motherboard for a few minutes, reconnecting it and it still showed screwed values on the capacity.

belec0, 2019-12-06:
Actually the best thing you can do is replace your battery by a new one, it would be dangerous to your switch if you keep using this one.

ShadowOne333, 2019-12-06:
> Actually the best thing you can do is replace your battery by a new one, it would be dangerous to your switch if you keep using this one.

I doubt it could be dangerous at all, it's rather incredibly annoying.
The only thing that happens is that the Switch cuts off my game at random percentage values.
Sometimes ~30%, other times around ~40%, but the battery does last the proper amount of time depending on the game that's being played.

And other times it takes ages to charge up to 100%, but I have been keeping close tabs on the capacity and voltage values, and they are always within the 3200k-4200k range.
It doesn't go below nor above those limits.

cannonball1911, 2019-12-07:
@ShadowOne333 

I have the same problem as you. What it "fixed" for me was to drain the battery to under 3600 mV and then loading it for a couple of hours. As long as I don't shutdown the Switch, the percentage display works fine.

I mean, it's not the best solution to let the Switch always in Standby but better than the Switch randomly shuts down

SciresM, 2019-12-31:
I would really prefer to not write battery code for now, because I am not a perfect dev and will make mistakes.

Also there's not really a good way to reimplement/mitm the stuff in the Bus sysmodule (and I'm not even really sure *how* to do what you're asking).

Will give more consideration if research is done to figure out how this would actually be done/if a spec is given?

closed 2019-12-31 by SciresM

ShadowOne333, 2019-12-31:
By specs you mean issue specs or system/battery specs?

SciresM, 2020-01-02:
By spec, I meant if a spec is given for how this would be implemented.