Atmosphere-NX/Atmosphere #703: Allow accessing the PMU

RSDuck, 2019-11-29:
## Feature Request

### What feature are you suggesting?
#### Overview:
If I tested it correctly the performance monitor is currently disabled for user mode applications. It probably shouldn't be enabled by default, but it would be great if could be enabled via a config entry.

#### Smaller Details: 
N/A

#### Nature of Request:  

I found [this](https://github.com/Atmosphere-NX/Atmosphere/blob/1e5fcff2422e86d220b6fde126b693ba3e31f155/exosphere/src/bootconfig.h#L28), so I guess there is already support for it from the OS side, though as I'm not familiar with it, so I don't know the full circumstances and ramifications.

### What component do you feel this would best fit within?
- [Exosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Fully-featured custom secure monitor.</br>

I think?

### Why would this feature be useful?
As a homebrew developer it would be great for profiling/messing around.

SciresM, 2019-11-29:
Oh, sure, I'm happy to add this.

What does the PMU actually do/what's the use case?

Just for my own curiosity -- this'll make it into either 0.10.0 or 0.10.1.

WerWolv, 2019-11-29:
The PMU consists of a bunch of counters and events that increment these counters. Here are some of the things that can be counted on a A9 chip (there's probably some more on the switch's A57): http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0388e/BEHEDIHI.html
Super interesting for things like Cache and TLB misses, branch counting, etc.

RSDuck, 2019-11-29:
to go really into the details, I'm optimising the 2D GPU renderer of melonDS with neon. And I'm currently stuck in a situation where there's hardly any information available, namely whether data dependencies are tracked by neon lane or by neon register. This is important because otherwise the following palette lookup code would be a disaster, since the ld2 instruction has a 6 cycles latency (minus the address calculation that's ~16*4 cycles wasted per loop iteration):

```
ubfx w0, %[tiledata], #0, #8
add x0, %[palette], x0, lsl #1
# 15-bit color
ld2 {v0.16b, v1.16b}[0], [w0]
ubfx w1, %[tiledata], #8, #8 
add x1, %[palette], x1, lsl #1
ld2 {v0.16b, v1.16b}[1], [w1]
# ...
```

I'm probably overthinking and this, but I still want to know whether stalling occurs or not.

SciresM, 2019-11-29:
Cool stuff. Very much interested in adding this. I'm expecting 0.10.0 within the next two or three weeks.

If you're building atmosphere locally or able to, message me on discord and I can tell you how to enable this yourself in exosphere in the meantime. Or I can provide you a build myself on Sunday or Monday.

RSDuck, 2019-11-29:
> If you're building atmosphere locally or able to, message me on discord and I can tell you how to enable this yourself in exosphere in the meantime. Or I can provide you a build myself on Sunday or Monday.

I have a math and a physics exam coming up, so I can wait for the official release.

closed 2019-12-07 by SciresM

RSDuck, 2020-01-09:
If anyone is interested here are my results to the problem described: https://community.arm.com/developer/ip-products/processors/f/cortex-a-forum/44706/partial-register-dependency-neon

commit 7e6ff1f3 (2019-12-07): exosphere: support enabling usermode pmu regs (closes #703)