Atmosphere-NX/Atmosphere #704: Seperated Atmosphere folders per sys/emuMMC instances

FusionFernia0000, 2019-12-01:
## Feature Request

### What feature are you suggesting?
#### Overview:
- To add seperated atmosphere folders for each sysMMC & emuMMC instances (that can contain different LayeredFS / kip / exefs / nxthemes patches / saves redirection / or even an entirely different version of Atmosphere ) so as not to interfere with other eMMC instances.

Note: This is mentioned in Atmosphere 0.9.0's changelog (Quote: "In addition, some niceties (e.g. having a separate Atmosphere folder per emummc instance) still need some thought put in before they can be implemented in a way that makes everyone happy."), but other than this, there is no other information to be found.

#### Smaller Details: 

- For sysMMC, it can either continue (or by default) to grab /sd_root/atmosphere/ folder for use, or add an extra config line to change the location of it. 
- As for emuMMC, the location of the atmosphere folder may be configured in /sd_root/emuMMC/emummc.ini (e.g. atmosphere_path=emuMMC/RAW1/atmosphere)
(hekate can also add support to change this line to achieve a fully configured environment without manually modifying emummc.ini)

#### Nature of Request:  
- Addition

### What component do you feel this would best fit within?
- [Fusée](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Atmosphère's custom bootloader.</br>
- [**Thermosphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Atmosphère's emuNAND implementation.</br>

*I'm not sure where to fit this in, since I'm not familiar with how the internals of Atmosphere work.

### Why would this feature be useful?
This can allow users and developers to quickly reboot to different instances of eMMC to test and use their desired controlled environment, instead of manually altering folders on SD card or even swapping to another card each time they want to use different patches.

*Words and solution are abruptly thrown together, my brain is not really functioning right atm, I can explain more details if anything is unclear.

noirscape, 2019-12-08:
To perphaps further elaborate, I think it would be the most useful to approach a setup like this:

- `/atmosphere/` -> Shared configuration folder for both sysMMC and emuMMC.
- `/sysmmc/` -> sysMMC specific configuration folder (configurable perphaps?).
- Dedicated emuMMC configuration folders according to the original post.

Going on from this, it would be the most handy as well to make `/atmosphere/` the _default_ configuration. For example, if I specify the mitm key in the `/atmosphere/` folder for the HBL to be `R`, but set it in the `/sysmmc/` folder to `L`, the `/sysmmc/` configuration would take precedence.

This approach also indirectly helps to separate ROM hacks, custom sysmodules and similar from the Atmosphere sysmodules (by virtue of placing them in the specific folder for the eMMC one chooses to boot) and makes updating Atmosphere less of a hassle.

That's my input on the matter.

Alexander3a, 2020-10-03:
+1 i woud really like that since i dont like swaping out the folder every time i wanna switch from emuMMC to sysnand to play online with cfw i woud really like it if u coud make a config file for that not just the one Atmosphere folder so its easy to change out other folders too.

i-lost-my-bagel, 2023-12-29:
I'd love this feature too. My solution for the past 2 years has been to use TegraExplorer as my bootloader and I wrote a script to move around some folders to switch between sysmmc and emummc and then boot atmosphere. [Here's the script and some instructions if anyone is interested.](https://github.com/i-lost-my-bagel/AtmosphereFolderSwitch)

metroid-maniac, 2024-10-03:
+1 for this issue. It seriously undermines the purpose and usefulness of having a separate sysMMC and emuMMC, or multiple emuMMCs, if the same sysmodules and other mods are loaded into all of them.

deepspaceaxolotl, 2024-11-03:
Any progress on this? It would be really helpful for differentiating emuMMC and sysCFW, making sure themes don't apply to the latter and minimising risk and such while still enabling swapping between them easily.

impeeza, 2024-11-03:
> Any progress on this? It would be really helpful for differentiating emuMMC and sysCFW, making sure themes don't apply to the latter and minimising risk and such while still enabling swapping between them easily.

What are you talking about?

sysCFW uses `SDMC:/Nintendo` folder.  and any EmuNAND uses the folder definied on the EmuNAND settings file.

deepspaceaxolotl, 2024-11-03:
Ah, apologies, I understood this issue being open as meaning that one can't have separate Atmosphere folders and thus can't install different themes or homebrew for emuNAND and sysCFW.

Random-Penguin, 2025-11-24:
+1, but I was maybe thinking a flag in the flags folder for the sysmodule / mod

(yes ik this is one year old)