Atmosphere-NX/Atmosphere #705: Atmosphere doesn't boot

ravi1913, 2019-12-05:
## Bug Report

### What's the issue you encountered?
Atmosphere does not boot, I only get a black screen when I try to inject the fusee-primary.bin file.

### How can the issue be reproduced?
I updated one of the games by accident, and it crashed almost instantaneously. It's a new game (Pokemon Sword), so I'm assuming it isn't supported as of yet. Now, my switch does not turn on. RCM mode works, but I am unable to get atmosphere to boot from both fusee and Hekate.

### Crash Report
N/A

### System Firmware Version

9.0.1</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee and hekate
- Official release or unofficial build: 
     - Kosmos' distribution of Atmosphère 

- Do you have additional kips or sysmodules you're loading:No
- Homebrew software installed: [ HBappstore, Tinfoil, Goldleaf]

### Additional context?
N/A

Skonikol, 2019-12-05:
You already have 9.1.0, wait for cfw update.

noirscape, 2019-12-05:
Piracy is not supported by the Atmosphere developers I'm afraid.

fincs, 2019-12-05:
Please try with a [completely unmodified copy of Atmosphère](https://github.com/Atmosphere-NX/Atmosphere/releases) without 3rd party piracy or hostile addons. If the problem persists we can reopen this issue.

closed 2019-12-05 by fincs

belec0, 2019-12-05:
the same thing happened to me i acidentally updated the system (it was not your pokemon, but the pop up appeared where you tried to open, happened with me on new super mario u deluxe), and Skonikol is right, if you updated the syste, you'll have to wait for the cfw update. What may not take so long...