Atmosphere-NX/Atmosphere #706: Blank screen when launching Atmosphère 0.9.4

jimbiano, 2019-12-05:
Using an unpatch switch but my kid accidentally updated the switch firmware today to 9.1.0. After system restart, unable to launch Atmosphere either directly through TegraRCMGUI or Hatake 5.0.2. When using TegraRCMGUI, it simply shows a blank screen with no logo. When using Hatake, the system simply restart Hatake again. Anyone encountering similar issue? Any quick help?

fincs, 2019-12-05:
Atmosphère is not currently compatible with 9.1.0. An update that corrects this issue is expected soon, please wait patiently until that happens.

jimbiano, 2019-12-05:
Thank you fincs. Can't wait for the update!

thedax, 2019-12-05:
@fincs , might it be helpful to pin one of these 9.1.0 issues to make it more visible and clear that Atmosphère has not been updated yet? Pinning an issue puts it in its own little section above the other issues at the top of the screen

Perhaps rename the issue "Atmosphère does work on 9.1.0 yet" as well.

fincs, 2019-12-05:
@thedax thank you for your suggestion; which has now been applied. All further issues reporting AMS not working on 9.1.0 will be treated as a duplicate of this one.

thedax, 2019-12-05:
You're welcome. :)

CheshireRose247, 2019-12-06:
please do not take this as rushing you, as I understand it can take time, but any guess when atmosphere update will be available? I accidentally updated my switch so just curious. but please take your time. sorry if any of that sounds rude i don't mean to come across as impolite in any way ^^'

jimbiano, 2019-12-06:
Other sites have published the required master key change for CFW to work due to 9.1.0. No rush but we hope that the dev quickly have Amos updated! Thank you.

haveamission, 2019-12-06:
Is it possible to update the master key on one's own without too much work? I'm a developer and not afraid to modify stuff as long as it's not an intensive process that'll take me hours (in which case I'll wait a few days for a patch)

kminz91, 2019-12-06:
Hello .
My switch is in 9.1.0 (yes i'm so sad xD)
My last firmware is 9.0.1 , it is dangerous to downgrade 9.1.0 in 9.0.1 ?
It's possible with "Choix du Jour"?

SciresM, 2019-12-06:
Commenting here to keep people updated.

The following is a list of the action items remaining before an 0.10.0 pre-release: 

❌ Finish ams_mitm refactor.
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;❌ Implement automatic backup behavior.
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;❌ Implement SaveData redirection.
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;❌ Implement HBL HTML redirection.
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;✔️ Implement Emummc Nintendo folder redirection.
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;✔️ Implement DirectoryRedirectionFileSystem
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;✔️ Implement SubDirectoryFileSystem
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;✔️ Implement FileSystemInterfaceAdapter (and File/Directory adapters).
❌ Add support for 9.1.0.
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;❌ Add new master key to exosphere.
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;❌ Add new master key to fusee-secondary.
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;❌ Change sept to derive new master key.
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;❌ Create new ams::TargetFirmwareVersion, update libstrat + exosphere to use it.
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;❌ Update emummc subrepo to commit that supports 9.1.0 (may require m4xw to update emummc).

I'm hoping to get the 0.10.0 pre-release out by Sunday-ish.

kminz91, 2019-12-06:
Hello SciresM? can you says If 9.1.0  burn More Efuses 9.0.1?

ZachyCatGames, 2019-12-06:
> Hello SciresM? can you says If 9.1.0 burn More Efuses 9.0.1?

Not Scires, but 9.1.0 burns 12, whereas 9.0.x burns 11. So yes it burns more

fincs, 2019-12-06:
Burn 🔥 

Locking this thread to avoid tangential discussions.

closed 2019-12-07 by SciresM