Atmosphere-NX/Atmosphere #708: Just Updated Atmophere, Error Desc: Data Abort (0x101)

kriemhild3ds, 2019-12-08:
hello, first time updating Atmosphere, I´m doing something wrong?
I put the new files on the sd, and I´m using the new fusee-primary.bin in TegraRCM
I tried on 2 different switch and I got the same error, what should I do!? please helpp

kriemhild3ds, 2019-12-08:
"Atmosphere can now display a fatal error screen significantly earlier in the boot process, if things go wrong early on."

what does this mean? I´m getting this screen :(

SciresM, 2019-12-08:
Please use the issue templates in the future. I've got a report of this elsewhere, though -- it looks like a custom theme triggered problem.

Delete your custom theme for a quick fix, or wait for a bug hotfix.

kriemhild3ds, 2019-12-08:
Just unistalled the custom themes on my switch and I got the same error :(, I put the old files and it works normally
the other switch doesnt have any custom themes installed and its on 9.1.0, and I get the same error :( I´m sorry x.x

SciresM, 2019-12-08:
Custom theme files are at /atmosphere/contents/0100000000001000. You may not have deleted the right folder.

kriemhild3ds, 2019-12-08:
thanks it works now, that was the problem I was unsinstalling the theme on NXThemes Installer and I had to do it deleting the folder you just said, thank you so much!!

kriemhild3ds, 2019-12-08:
on the console on 9.1.0 I keep getting a crash after 10 seconds on CFW, I run HB through the album and everything works perfect, but after 10 seconds it crashesss x.x

kriemhild3ds, 2019-12-08:
[image]

😭

SciresM, 2019-12-08:
Delete the custom system module ^^

poketradecenter, 2019-12-08:
hi please can you help me im unsure of what to do i downloaded all the files for 9.1 but not sure where to put them on my sd card ?? could you please explain to me what to do

commit 889ea5e5 (2019-12-08): Fix memory corruption in romfs build (closes #708)
closed 2019-12-08 by SciresM