Atmosphere-NX/Atmosphere #709: Atmosphere v0.10 Will Not Boot in Firmware 8.1.0

dothackjhe, 2019-12-08:
## Bug Report

### What's the issue you encountered?

Tried updating to the latest version of Atmosphere (version 0.10), but found it unable to boot. Never had this issue using Atmosphere v0.94.</br>

### System Firmware Version

8.1.0</br>

dothackjhe, 2019-12-08:
I reverted to using the files for Atmosphere 0.9.4 and everything worked perfectly again.

friedkeenan, 2019-12-08:
I can boot successfully on 8.1.0. Disable any sysmodules you might have running and please give more information than "I can't boot"

dothackjhe, 2019-12-08:
I do not know how to put it differently. 

Initially, it got stuck after the Kosmos splash screen. My second attempt, on the other hand, loops me back to the screen where I get to choose to boot the system into CFW (Sysnand) [I do not know how this part of the system is called, TBH].

SciresM, 2019-12-08:
This sounds like "hekate doesn't support 9.1.0 yet" to me.

closed 2019-12-08 by SciresM

dothackjhe, 2019-12-08:
Sounds likely.

dothackjhe, 2019-12-09:
The latest update of Hekate addressed the issue for me.