Atmosphere-NX/Atmosphere #711: NEED HELP PLEASE WITH NEW ATMOSPHERE FOR 9.1

poketradecenter, 2019-12-08:
hi i just downloaded all the files for the new atmosphere could someone please help me on what i need to do where i need to put the files? 

i downloaded the atmosphere-prerelease-0.10.0-master-7e6ff1f3+hbl-2.2+hbmenu-3.1.1
extracted the files and pasted them to my sd card i then inserted my sd into my switch used tegra to inject the payload as i usually do with hekate 
it boots up i click launch 
i am then given 3 options sysnand / emummc and stock 
i clicked on emummc as i used too 
when i do that it takes me back to the start again 
am i doing this right?
any help would be greatly appreciated thankyou

SciresM, 2019-12-08:
Hekate doesn't support 9.1.0 yet.

Also, please do not create issues without using the issue templates.

closed 2019-12-08 by SciresM

poketradecenter, 2019-12-08:
sorry im unsure of this just created my account today can you help me on tegra what should i boot as i sually use hekate