Atmosphere-NX/Atmosphere #713: Fatal Error While Rebooting Atmosphere

AlinMGupta, 2019-12-08:
## Bug Report

### What's the issue you encountered?

When Rebooting atmopshere using the reboot to payload NRO in the homebrew menu, i got the error:
A Fatal Error occured when running Atmosphere.
Title ID: 010041544d530000
Error Desc: std::abort() called (0xffe)
Press Power to reboot.

Before hand i installed the FTPD service which i tested and worked. 

### Crash Report

Fatal error file saved as a .bin file so cant upload it

### System Firmware Version

Switch 9.1.0</br>

### Environment?

- Hekate, Using the FuseePrimary.bin file:
- Official release or unofficial build: 
  - Atmopshere Official release version 10
 

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]
I Can boot to Hetake by holding the power button, but using fusee primary sends me back into the loop.

I can also boot to stock by removing the SD card as trying to do it through Hetake doesnt do anything.

When i press power it gives me the same error.

NWPlayer123, 2019-12-08:
for the bin, if you're on Windows, you can right click, send to -> zipped folder and GitHub will let you upload the zip
[image]

SciresM, 2019-12-08:
What exactly did you do to trigger this problem?

AlinMGupta, 2019-12-08:
Thanks NWPlayer123

ScieresM,
I used the reboot to payload Button in the Homebrew menu.

AlinMGupta, 2019-12-08:
Heres a strange thing. I installed ULaunch before using the button, and deleted it afterwards incase that was the problem. but the folder keeps appearing again, when plugged into the switch.

NWPlayer123, 2019-12-08:
are you using FAT32 or exFAT?

AlinMGupta, 2019-12-08:
I am using FAT32.

SciresM, 2019-12-08:
Cannot reproduce. I successfully used the reboot to payload with no problems twice in a row just now.

closed 2019-12-08 by SciresM

AlinMGupta, 2019-12-08:
> Cannot reproduce. I successfully used the reboot to payload with no problems twice in a row just now.
Can you provide a fix for my switch? i cant use home-brew at this rate.

SciresM, 2019-12-08:
1) Make sure you're using a clean copy of atmosphere from the release branch

2) What do you even need reboot to payload for? Atmosphere provides a mitm so that just selecting reboot from the power menu does a reboot to payload, to the NRO is basically completely superfluous.

AlinMGupta, 2019-12-08:
> 1. Make sure you're using a clean copy of atmosphere from the release branch
> 2. What do you even need reboot to payload for? Atmosphere provides a mitm so that just selecting reboot from the power menu does a reboot to payload, to the NRO is basically completely superfluous.

1. I am, should i just reinstall?
2. Thanks, i did not know this, will take note for the future.

SciresM, 2019-12-08:
I would try re-installing -- I can't reproduce your problem at all though

AlinMGupta, 2019-12-08:
> I would try re-installing -- I can't reproduce your problem at all though
Okay thanks.

AlinMGupta, 2019-12-09:
> I would try re-installing -- I can't reproduce your problem at all though

Sorry to bother you, but after some testing, i have linked the problem to ULaunch. There is already an issue open on their feed, but maybe warn people that something as widely used as it causes a fatal error on startup.