Atmosphere-NX/Atmosphere #716: 0.10.0 failed to boot. Error 2003-0008 (0x1003)

DitDitGitHub, 2019-12-08:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

An error code on the boot screen after trying to update Atmosphere. Error code is: 2003-0008 (0x1003) and the program is 010000000000100C. I am on FW 9.0.1.

### How can the issue be reproduced?

1. Install Atmosphere 0.10.0 files over 0.9.4 files.
2. Using the fusee-primary payload, boot into Atmosphere, either inside or outside of Hekate.
3. Receive the error.

### Crash Report

[01575795127_010000000000100c.log](https://github.com/Atmosphere-NX/Atmosphere/files/3936007/01575795127_010000000000100c.log)

### System Firmware Version

9.0.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
      TegracmGUI and Hekate.
- Official release or unofficial build: 
      Official release 0.10.0
- Do you have additional kips or sysmodules you're loading:
      No, I have uninstalled them.
- Homebrew software installed: [ AmiiSwap, AppStore, Checkpoint, Goldleaf, HekateBrew, Incognito, KipSelect, LDNMitm Config, MMLinx, NX-Activity-Log, NX-Shell, SwitchPresence,Rewritten, Sys-Clk Editor, Tinfoil, VGEdit, FTPD, Retroarch.]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

DitDitGitHub, 2019-12-08:
I can confirm that downgrading to Atmos 0.9.4 works.

SciresM, 2019-12-08:
This is an error due to custom sysmodules.

DitDitGitHub, 2019-12-08:
Yeah, I just realized that I didn't delete the sysmodules. I'm gonna try again.

DitDitGitHub, 2019-12-08:
I've deleted the sysmodules from the titles folder, still doesn't work.

SciresM, 2019-12-08:
you need to delete from the contents folder. Atmosphere no longer uses the titles folder.

pixel-stuck, 2019-12-08:
titles was moved to contents now. Read the release notes; a small utility was added to move the titles foleder's contents into contents, so you'll have to delete the contents folder.

DitDitGitHub, 2019-12-08:
Ok

DitDitGitHub, 2019-12-08:
After deleting the folders from content, I don't get to the Nintendo Switch screen.

SciresM, 2019-12-08:
You don't want to delete every folder -- some of them are atmosphere's system modules. Delete folder, extract clean latest release zip.

DitDitGitHub, 2019-12-08:
Ok

DitDitGitHub, 2019-12-08:
Thank you, it has worked ! :)

closed 2019-12-08 by DitDitGitHub