Atmosphere-NX/Atmosphere #717: Fatal Error when booting Atmosphere

melgarr, 2019-12-08:
Title ID: 010041544d530000
Error Desc: std::abort() called (0xffe)
[report_0000000005e8a81c.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/3936109/report_0000000005e8a81c.bin.zip)

Formatted SD card and using clean install from latest release. After injecting fusee primary, atmosphere logo appears, then the smaller logo and then the fatal error screen pops up. This happens every time, haven't been able to boot successfully.

SciresM, 2019-12-08:
Are you able to get on discord? Your issue is a weird one (a file doesn't exist that atmosphere asserts does). Happy to debug with you if you can.

melgarr, 2019-12-08:
Yeah, that does seem suspicious...I'll hop on in a minute

SciresM, 2019-12-08:
...Does your SD card not have enough space remaining to create a 1 MiB file(?)

SciresM, 2019-12-08:
Wait, the backups are only 0x8080 bytes, super weird.

melgarr, 2019-12-08:
It's a 256gb samsung card, I just formatted it so it's empty besides the atmosphere files. Format is exFAT currently

SciresM, 2019-12-08:
>exFAT

Format it as FAT32.

fwiw the error in your report is that Atmosphere fails to open the file to save automatic NAND backups because the file doesn't exist -- suggests to me that file creation is failing, somehow.

melgarr, 2019-12-08:
FAT32 fixed the issue, thought atmos supported exFAT? Got to boot and see the homebrew menu so all seems well.

melgarr, 2019-12-08:
Quick question, am I able to drop my Nintendo folder back into the SD with no issues? Or since it was saved in exFAT am I out of luck? Mainly care about my pokemanz saves.

SciresM, 2019-12-08:
Atmosphere does support exFAT -- I have no idea why backup file creation would fail for you.

SciresM, 2019-12-08:
You should be able to drop a backed up N folder in just fine

closed 2019-12-08 by SciresM

melgarr, 2019-12-08:
Yeah, that is very weird. Going to reformat back to exFAT and see if it crashes again for science.

tsutsu, 2019-12-08:
I had the same error; it was fixed by running Hekate's "Unset archive bits" tool. FYI!

melgarr, 2019-12-08:
Yeah, formatting back to ExFat breaks it again. That is interesting though, I was looking into your solution and I am using MacOs Drive utility to format my SD card, I suppose it does it in a way that doesn't play nice with the switch (DS_STORE files, archive flags, etc). I'll give the hekate tool a try and report back in case anyone else has a similar config/issue

Jabo-SCO, 2019-12-09:
The issue is not the ExFat format, it's the archive bit. Use Hekate and unset archive bit and you can use it again.

(My SD is formatted by MacOS Disk Utility in ExFat)

aldovaleriani, 2019-12-13:
Hey guys di you figure out what was the problem cant find a way around it. i keed getting the same error

jcmac687, 2019-12-14:
> The issue is not the ExFat format, it's the archive bit. Use Hekate and unset archive bit and you can use it again.
> 
> (My SD is formatted by MacOS Disk Utility in ExFat)

Made an account just to comment, this totally fixed my issue after hours of troubleshooting

burgerkingrei024, 2019-12-16:
> The issue is not the ExFat format, it's the archive bit. Use Hekate and unset archive bit and you can use it again.
> 
> (My SD is formatted by MacOS Disk Utility in ExFat)

I tried this and now whenever I go to boot emusys it goes to a black screen, any help?