Atmosphere-NX/Atmosphere #718: (9.1.0) Fusee Primary boots into black screen

RowanHall, 2019-12-08:
## Bug Report

### What's the issue you encountered?

After injecting `fusee-primary.bin` (latest, from /releases) the backlight turns on, but the lcd displays black. If the power button is pressed, the backlight turns off, then after a short delay turns back on. (it should be noted that the windows disconnect sound plays from my laptop, and TegraRCMGUI (v2.6) displays NO RCM.

This black screen does not go away untill the switch is force shut-down by holding the power button.

### How can the issue be reproduced?

1. Boot switch into RCM.
2. Inject fusee-primary

### Crash Report

The folder crash_reports does not exist.

### System Firmware Version

9.1.0</br>

### Environment?

fusee (v latest, atmos release 0.10.0)

### Additional context?

Note the SD card is inserted.

Webbeh, 2019-12-08:
Exactly the same here.
SD card formatted as exfat.

RowanHall, 2019-12-08:
I don't get any splashes, it just directly goes to black

Webbeh, 2019-12-08:
Launching it with Fusee-Primary gets splashes and blackscreen.
Launching it from Hekate just reboots straight to hekate.

RowanHall, 2019-12-08:
mine goes directly to black with the effects i provided above, weather or not i launched directly or through hekate.

SciresM, 2019-12-08:
@RowanHall this sounds like a "your SD card doesn't have fusee-secondary" problem.

Extract clean release zip to SD?

death2all110, 2019-12-09:
Getting a black screen when booting 0.10.0 from Hekate 5.1.0. Nothing in fatal_errors, no crash reports directory. SDCard is in exfat format.

ELY3M, 2019-12-09:
can you try deleting sept folder and replace with new sept folder from atmosphere zip?

death2all110, 2019-12-09:
I think i may have attempted that , but I'll give it a shot after work. Thanks.

death2all110, 2019-12-10:
Ok. so tried deleting sept and copying again. same issue.

SciresM, 2019-12-10:
@death2all110 send picture of the root of your SD card, and of your atmosphere folder?

death2all110, 2019-12-10:
Sure. Here you go:

Also, i guess its not really a black screen but its actually turning the switch off.

[image]
[image]

SciresM, 2019-12-10:
>macOS

It's almost certainly an archive bit problem.

Use hekate's archive bit fixer?

death2all110, 2019-12-10:
Son of a b*tch. 

I am so sorry. I completely forgot about all that.

closed 2019-12-10 by SciresM

relativenukes, 2019-12-13:
I have the exact same issue as RowanHall.
Switch version 9.1.0
Everything is on the newest update
When I try to boot the Fusee primary payload, the screen stays on and black. Pressing the power button turns screen off temporarily.
Any help would be appreciated, thanks.

burgerkingrei024, 2019-12-16:
> > macOS
> 
> It's almost certainly an archive bit problem.
> 
> Use hekate's archive bit fix

Using the bit fix was what caused the issue for me

TheDiggleGod, 2020-02-01:
It seems I have the exact same problem as described in the OP. 

Something he didn't mention that I think might be helpful, is when I try to launch through Hekate's CFW (emuMMC) some text appears saying "unknown pkg1 version. failed to launch HOS." And when I inject fusee-primary, either via hekate or TegraRCMGUI, the screen cuts off, and it'll turn out, but the screen is still black.

I've tried the archive bit fix and reinstalling hekate and atmosphere, but I'm still having the problem.

tklitschi, 2020-05-10:
I had the same Problem after updated my atmosphere. I'm using FAT32. After updating my atmosphere directly via PC and updated hole boot loader I got it kinda working again, but my switch didn't found any game. 
After formatting hole SD Card I got it working again.

hscuddera, 2020-06-24:
My advice when it comes to issues like these is to redownload all the standard SD setup type stuff. Then go through your sd card and replace the maximum u can without getting rid of stuff that looks different. Then try to boot switch.

Zaneoi, 2020-09-29:
Im having a similar issue to, i can boot to emummc through hekates configs but i cant boot using fusee primary from payloads or by using tegrarcmgui on my pc. Every time i try its just a splash screen then black screen with a backlight.

keinushi, 2021-02-09:
Installed atmosphere 0.18.0 and it booted right up. Been having issues with 0.16.2 since I installed it, this seemed to fix everything.

makermelissa, 2021-12-20:
> > macOS
> 
> 
> It's almost certainly an archive bit problem.
> Use hekate's archive bit fix

I'm new to this and I was having the same issue after formatting to ExFat on MacOS. This fixed it for me as well. Thanks for the suggestion.