Atmosphere-NX/Atmosphere #723: Fatal Error: Unable to read (fusee-secondary.bin)

RoyallyOblivious, 2019-12-08:
I get this error message pop up, i don't know what to do and I've exalted everything I could come up with.
[image]

SciresM, 2019-12-08:
Make sure the release zip is extracted to your sd card...?

RoyallyOblivious, 2019-12-08:
I have it in my SD card, but it won't load it, I'm not really good at doing this, I'm new to this

SciresM, 2019-12-08:
What does the atmosphere folder on your SD look like? Post a screenshot.

RoyallyOblivious, 2019-12-08:
[image]

shchmue, 2019-12-08:
move everything from the `sd` folder onto the root of your sd

SciresM, 2019-12-08:
You need to extract to the root of your sd card, not a "sd" folder.

closed 2019-12-08 by SciresM

RoyallyOblivious, 2019-12-08:
I already did that too

RoyallyOblivious, 2019-12-08:
[image]