Atmosphere-NX/Atmosphere #724: Atmosphere crashes after the switch logo when loading emunand

flopglop, 2019-12-08:
## Bug Report
### What's the issue you encountered?

After installing atmosphere and loading fusee-primary.bin atmosphere crashes after the switch logo with the error code 0x149402 (2002-2634) when booting from Emunand.

### How can the issue be reproduced?

1. Boot switch into RCM.
2. Enable Emunand.
3. Inject (or chainload) fusee-primary.bin.

### Crash Report

https://gist.github.com/flopglop/1d76c88be710cff4e621c50fdd603c15

### System Firmware Version

8.1.0 Emunand
9.1.0 Sysnand

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
This issure occurs both when loading fusee-primary.bin or chainloading it.

- Official release or unofficial build: 
Official release version 0.10.0.

### Additional context?

The switch boots to recovery mode, and I use a partition based Emunand.

EdwardElric64, 2019-12-08:
Same issue here. Loading hekate and trying to boot cfw from SysNAND just crashes back to the payload

Koha1297, 2019-12-08:
> Same issue here except I get a different error code. Loading hekate and trying to boot cfw from SysNAND just crashes back to the payload

Hekate doesn't support 9.1.0 yet. Booting SysNand or Stock will revert back to Hekate.
https://github.com/CTCaer/hekate/issues/314

SciresM, 2019-12-08:
This error code means you don't have nogc enabled and are booting into a version with incompatible gamecard firmware.

Turn on nogc.

closed 2019-12-08 by SciresM