Atmosphere-NX/Atmosphere #726: Fatal error when booting up

MegaSuperab, 2019-12-08:
## Bug Report

### What's the issue you encountered?

Booting up Atmosphere using fusee-primary causes a fatal error to occur. The only changes made to Atmosphere were changing the applet override key to R and changing the cheat enable key to X.

### How can the issue be reproduced?

1. Inject fusee-primary.bin
2. Boot up Atmosphere

### Crash Report

A fatal error occured in this issue, so I will provide the bin file here.  [report_00000000175078b0.zip](https://github.com/Atmosphere-NX/Atmosphere/files/3936553/report_00000000175078b0.zip)

### System Firmware Version

9.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusèe
- Official release or unofficial build: Official release 0.1.0 prerelease
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: I'm honestly too lazy to list them all in text so here's an image instead:
[image]

### Additional context?

- Additional info about your environment: I'm using FAT32 this time.

SciresM, 2019-12-08:
The crash is happening due to an assertion failure in romfs building code.

Do you have a custom theme? I suspect it may be corrupted.

Could you post your /atmosphere/contents folder?

MegaSuperab, 2019-12-08:
After a quick test I found out that uLaunch was causing the issue. Uninstalling it makes Atmosphere work correctly. Thank you for your help.

closed 2019-12-08 by MegaSuperab