Atmosphere-NX/Atmosphere #727: fsmitm remains always active

Kronos2308, 2019-12-08:
## Bug Report

fsmitm remains active even if fsmitm.flag does not exist

### What's the issue you encountered?
These are the files I have
"atmosphere\contents\0100000000001000\romfs\lyt\Flaunch.szs"
"atmosphere\contents\0100000000001000\romfs\lyt\ResidentMenu.szs"
"atmosphere\contents\0100000000001000\romfs\lyt\Notification.szs"
and 
"atmosphere/contents/0100000000001000/fsmitm.flag" 
should enable and disable the fs
also try "atmosphere/contents/0100000000001000/flags/fsmitm.flag"

even if you delete the .flag it always remains active

### How can the issue be reproduced?
### Crash Report
### System Firmware Version
9.1.0
### Environment?
fuse primary

### Additional context?

SciresM, 2019-12-08:
This isn't a bug, it's intended behavior.

Qlaunch should always be redirected.

closed 2019-12-08 by SciresM

Kronos2308, 2019-12-08:
ok

SciresM, 2019-12-08:
(this is, I believe, also how it worked on 0.9.4.)

Kronos2308, 2019-12-08:
but if you use custom themes is a problem ,
cant we have a flag to disable ?

SciresM, 2019-12-08:
Rename the romfs directory...?

The flags were complicated for no real benefit.

Kronos2308, 2019-12-08:
ok, thanks
i will rename then