Atmosphere-NX/Atmosphere #731: Incognito / PRODINFO / ams_mitm kips

AntHJ, 2019-12-08:
The console was officially updated with the latest fw 9.1.0

An emuMMC was then created which only the latest Fusee bin and AMS 10 will boot the system.

I then downgraded the emuMMC to 9.0.1 (tried 8.1 too) and still only the latest Fusee bin and AMS 10 will boot the system which is expected.

I tried to add the files needed to allow incognito to edit the Prodinfo but when I try to boot up it freezes. If I remove the kips/flags it boots up fine..

I assume something in the new atmosphere has changed or stopped this ability

How can I run incognito to edit the prodinfo?

Maybe add a feature where we can add a line ProdInfoAccess=true

SciresM, 2019-12-08:
Actively not supported.

Also, please use the issue templates.

SciresM, 2019-12-08:
Don't use incognito.

closed 2019-12-08 by SciresM