Atmosphere-NX/Atmosphere #733: atmosphere crash after tap on album icon

ELY3M, 2019-12-08:
## Bug Report
I have been working with my deaf friend setting up new atmosphere on his switch.  
when he tap on album icon to open hbmenu, it result in atmosphere crash.  

atmosphere crash when i tap on the album icon.  

### What's the issue you encountered?
atmosphere crash  

Describe the issue in detail and what you were doing beforehand. ] 
just tapping on album to open the hbmenu.   

Did you make any changes related to Atmosphère itself? not at all. no changes 
[ If so, make sure to include details relating to what exactly you changed. ] N/A
 

### How can the issue be reproduced?
just tapping on album icon to open hbmenu

### Crash Report
https://www.dropbox.com/s/07mcc2wc55z4ze1/fatal_reports.rar?dl=1

### System Firmware Version
9.1.0 

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
atmosphere was launched by latest fusee-primary.bin  
official build: atmosphere-prerelease-0.10.0-master-3c7ece2f+hbl-2.2+hbmenu-3.1.1

- Do you have additional kips or sysmodules you're loading: sys-ftpd
- Homebrew software installed: edizon

### Additional context?

SciresM, 2019-12-08:
"Do you have additional kips or sysmodules you're loading: sys-ftpd"

Remove sys-ftpd.

closed 2019-12-08 by SciresM

ELY3M, 2019-12-08:
Thank you!  that was it.  I deleted sys-ftpd and it did worked  :)