Atmosphere-NX/Atmosphere #734: Unable to identify package1

RoyallyOblivious, 2019-12-08:
I had posted earlier about the fusee-secondary file not being read, now it is saying that it is unable to to identify package1

SciresM, 2019-12-08:
@RoyallyOblivious it really sounds to me like you're not actually using the latest release fusee-primary + fusee-secondary. Please make sure you're using the latest release binaries...

RoyallyOblivious, 2019-12-08:
I am using Tegra to go into CFW, is that an issue?

NWPlayer123, 2019-12-08:
You need the latest payload, along with the latest Atmosphere release. I'd recommend
* backing up SD:/atmosphere and SD:/sept
* deleting those folders
* deleting your old Tegra atmo payload
* grabbing zip + fusee-primary from https://github.com/Atmosphere-NX/Atmosphere/releases/tag/0.10.0
* extracting the whole zip to your SD card to get the latest atmo/sept files
* using latest fusee-primary with Tegra

RoyallyOblivious, 2019-12-08:
@NWPlayer123 

What's atmo and I'm using the recent release of Tegra to load the CFW, I already have entered the RCM on the switch, but I'm having no luck

NWPlayer123, 2019-12-09:
latest version of Atmosphere. You need to update sd:/atmosphere, sd:/sept, and the xxx.bin file you use with Tegra *every time an update comes out*
[image]
[image]
[image]

RoyallyOblivious, 2019-12-09:
I got the atmosphere screen up and then it goes to the nintendo logo after it loads, is that normal, what else should i do? This is the farthest I have gotten, so now what else do I need to do?

NWPlayer123, 2019-12-09:
That's normal, after that it should just boot same as a normal Switch. Did it pop up with a crash log or make a file or something?

RoyallyOblivious, 2019-12-09:
@NWPlayer123 nope, so what do I do to access the home brew and to edit my game data?

closed 2019-12-09 by SciresM