Atmosphere-NX/Atmosphere #736: Error Code: 2162-0002 (0x4a2)

EnriqueJuarez918, 2019-12-08:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]<Error when booting up atmosphere>
[ Did you make any changes related to Atmosphère itself? ]<i installed the new  release to my sd card>
[ If so, make sure to include details relating to what exactly you changed. ]<extracted folder files onto the sd card>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]
<https://gist.github.com/EnriqueJuarez918/8e9d03fade9a5ebb523292c3292067de>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]

### System Firmware Version

9.1.0
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Fusee
- Official release or unofficial build: official build
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: [ * ] atmosphere

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2019-12-08:
Sounds like a custom theme problem. Does the error go away if you delete /atmosphere/contents/0100000000001000?

EnriqueJuarez918, 2019-12-08:
yes, it was the custom theme that was causing the problem. thank you

closed 2019-12-08 by SciresM

xmagcx, 2020-03-17:
Hi guys waiting to be fine I have the error 010000000000100f (attached log of the folder fatal report). As evidence it happens to me with the game super smash bros ultimate when starting both story mode or any other mode. What I have tried so far:
1- Just play the version without updates.
2- install the latest update
3- install different updates and nothing

Please I need your help I have tried and tried everything in my power

[01584409357_010000000000100f.log](https://github.com/Atmosphere-NX/Atmosphere/files/4345310/01584409357_010000000000100f
[image]
.log)

Juanperezc, 2020-04-09:
Same issue with the same game

Juan1893CR, 2020-04-11:
I have the same issue :(

tenchi19134, 2020-04-26:
Thx @SciresM. Worked for me.

thedoofmodern-backup, 2020-06-18:
For me. it dosen't give 0100000000001000(custom theme problem) but 4200000000000010 with error code 2168-0002 happens at boot

xmaurox9, 2020-06-18:
> 
> 
> Same issue with the same game

Did you solve it?

hexal6, 2020-08-15:
Work! great! deleting folder  -  /atmosphere/contents/0100000000001000
themes trouble!

Neules, 2020-11-30:
Same error here playing Mario Odyssey. It happens when leaving the Mario _movements guide_. As the game is done to go though this guide to get next stage it always crashes.

I don't have the (in)famous `0100000000001000` folder so, what else have you tried ppl?

Atmosphere ver: 0.14.3-master-28ceedb5
Firmware: 10.2.0

SciresM, 2020-11-30:
This issue is a closed one. Please open a new issue filling out the template with attached crash report, else seek support in the ReSwitched or Nintendo Homebrew discord servers.

Neules, 2020-11-30:
Sorry for that. Where can I find the crash error? It is located in `crash_reports` folder? Thank you very much and excuse me for the inconvenience.

SciresM, 2020-11-30:
If the error screen was blue with white text, the error log will likely be inside the `crash_reports` folder, yes, although it may be inside the `fatal_reports` folder.

If the error screen was black with white text, the error log will be a binary file inside the `fatal_errors` folder.

yaelboy36, 2021-05-16:
[01621137702_010000000000000c.log](https://github.com/Atmosphere-NX/Atmosphere/files/6488694/01621137702_010000000000000c.log)
I'm having this problem everytime I enter to atmosphere. I was playing fine some minutes ago and this appeared suddenly. After that, everytime I reboot it always shows this error.

D3adl0ck420, 2022-01-11:
anybody know why 0100000000001000 has errors? might work on a custom patch for it idk

Masamune3210, 2022-01-11:
please do not co-op existing issues, make your own next time
01000...1000 errors are home menu and usually caused by incompatible themes

SciresM, 2022-01-11:
(also don't make an issue for 0100000000001000, this is home menu and almost always user error. Seek support on discord first.)

CoolWillBR, 2022-05-24:
Hello, I have a problem with the program 0100000000000014, when I try to start the game in super nintendo online it gives this error, I went to add new games now but it still gives the same error, I already played the nintendo 64 online and it runs fine, and the firmware is 14.1.1 (atmosphere 1.3.1-master-62eb4d698)

armandoantipas, 2023-03-15:
> Sounds like a custom theme problem. Does the error go away if you delete /atmosphere/contents/0100000000001000?

You are a life saver man. I appreciate it.

Luis01Felipe, 2024-04-10:
> Sounds like a custom theme problem. Does the error go away if you delete /atmosphere/contents/0100000000001000?

Thanks dude it solved my problem, I was thinking to reset format the console, thank god this worked

armandoantipas, 2024-04-10:
That was exactly the issue. I deleted the custom themes and it booted
perfectly. I appreciate the answer. Have an awesome day.

TheVgam3r, 2024-10-26:
i'm currently having this error when i try to openMario & Luigi Brothership.

the game loads a bit then it crashes
[image]

here's the log file that it gives me
[
[01729920804_0100000000000014.log](https://github.com/user-attachments/files/17529146/01729920804_0100000000000014.log)
](url)