Atmosphere-NX/Atmosphere #737: Cant compile atmos with latest devkitpro

Ntemis, 2019-12-08:
https://paste.ubuntu.com/p/23CFSDdVNW/

SciresM, 2019-12-08:
Atmosphere requires libnx-master at the moment. If you wish to compile atmosphere, make sure you're using libnx-master and not libnx-stable.

(However, there's also not much reason to want to compile by yourself atm).

closed 2019-12-08 by SciresM

fincs, 2019-12-08:
[There's no such thing as "latest devkitPro"](https://devkitpro.org/wiki/Getting_Started#Setup). The current pre-release version of Atmosphère relies on various pre-release versions of components distributed and published by devkitPro; since we work in tandem. You can expect the final stable release of Atmosphère to be buildable with the stable versions of said components we will make available as soon as we're able to.