Atmosphere-NX/Atmosphere #738: Crash after booting primary fuse

DJCR4V3, 2019-12-08:
## Bug Report
error code: 2162-0002 (0x4a2)
program: 0100000000001000
firmware: 9.1.0 (atmosphere 0.10.0-master-ea49c2cc

### What's the issue you encountered?

after accidently accpecting system update my switch wouldnt boot my CFW i downloaded the current atmospere and fuse bin files. when i boot it crashes

### How can the issue be reproduced?
boot into hakate while using firmware 9.1.0. 
select primary fuse bin 
wait to boot and crash?
[01571987511_010000000000100d.log](https://github.com/Atmosphere-NX/Atmosphere/files/3937180/01571987511_010000000000100d.log)

### Crash Report

[01571987511_010000000000100d.log](https://github.com/Atmosphere-NX/Atmosphere/files/3937180/01571987511_010000000000100d.log)

### System Firmware Version

9.1.0
### Environment?

-hakate current version. atmosphere0.10 
### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2019-12-08:
Delete your custom theme by deleting the folder at /atmosphere/contents/0100000000001000

closed 2019-12-08 by SciresM

DJCR4V3, 2019-12-08:
i love you more than words can explain

sebas7iano, 2021-08-31:
> Delete your custom theme by deleting the folder at /atmosphere/contents/0100000000001000

man, if you want me to suck you dick just tell me bro. thank you so much for you help, now the system works fine!

Masamune3210, 2021-08-31:
....you really made a github account just to say that?

sebas7iano, 2021-08-31:
> ....you really made a github account just to say that?

Where is the problem? I was stucked and thanks to his message I was able to fix the bug? I wanted to thank him in a funny way, but if GitHub is yours and you want me to leve It say it. I will obviously do It ..............🤦🥱