Atmosphere-NX/Atmosphere #739: Edizon Issue

optimism247, 2019-12-08:
After installing Atmosphere 0.1.0, I am unable to access cheats from Edizon. 

Even after these cheats have been moved to "atmosphere/contents/cheats/", they cheats don't show up at all. 

Is this an issue related to Edizon or Atmosphere?

HookedBehemoth, 2019-12-08:
It's because EdiZon assumes they are still in the titles folder.
I opened a PR to fix compatibility with AMS 0.10.0+. Check for progress there or join EdiZon's server.
https://github.com/WerWolv/EdiZon/pull/98
https://discord.gg/qyA38T8

closed 2019-12-08 by SciresM

optimism247, 2019-12-08:
Thanks so much, Hooked. I appreciate that a lot. 

I assumed that it was most likely an Edizon issue, considering it hasn't been updated in quite some time. I'll jump over there now.