Atmosphere-NX/Atmosphere #740: Error 0x4A8 (2168-0002)

EdwardElric64, 2019-12-08:
## Bug Report

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
Attempting to boot into atmosphere after updating to firmware 9.1.0 and atmosphere 0.10.0 crashes right after showing the switch logo
[ Did you make any changes related to Atmosphère itself? ]</br>
No
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>
Install latest firmware and atmosphere, load fusee-primary
### Crash Report

https://gist.github.com/EdwardElric64/059d3c4b4edcb004b85e7a29f57b4343

### System Firmware Version

9.1.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee
- Official release or unofficial build: 
  - Official, 0.10.0
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: Edizon, Goldleaf, incognito, kosmos toolbox/updater, ldnmitm_config, Lockpick, mercury, NXthemes_installer, Payload_launcher, smw, tinfoil, freebird

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2019-12-08:
Per crash report, this is an emuiibo error, not an atmosphere error.

closed 2019-12-08 by SciresM

x302justice, 2020-02-09:
Attempt to install homebrew launcher Nsp on goldleaf using emummc. Issue only happens on hbl, goldleaf nsp installs. Firmware 9.1.0