Atmosphere-NX/Atmosphere #741: EdiZon Cheats crash on AC Rebel Collection

Nomercyman1, 2019-12-09:
What's the issue you encountered?
Opened EdiZon to use its Cheats (RAM Editor) and the system crashed to a blue error screen. Only happens on this game, other games work fine (SwSh used to test as well).

How can the issue be reproduced?
Open EdiZon, with game open. Go to Cheats.

Crash Report
https://gist.github.com/Nomercyman1/92bdf4ee30139bc44fc52c8e15924bf5

System Firmware Version
0.9.1

Environment?
What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
fusee

Official release or unofficial build:
0.10.0 Atmosphere from site (latest build)

Do you have additional kips or sysmodules you're loading:
No

Homebrew software installed: [ * ]
EdiZon (latest build), Checkpoint, App Store

Additional context?
Additional info about your environment:

SciresM, 2019-12-09:
Make sure you're using the latest binary -- some cheat stuff was fixed.

Also, this is an edizon issue, not an atmosphere problem.

closed 2019-12-09 by SciresM

Nomercyman1, 2019-12-09:
I have spoken with people about EdiZon and updated everything to the latest. It is a specific issue with dmnt cheats in Atmosphere, I believe. As I said, Pokemon Shield cheats work fine

SciresM, 2019-12-09:
Atmosphere code cannot cause ediZon to data abort.

Nomercyman1, 2019-12-09:
[image]

SciresM, 2019-12-09:
```
Atmosphère Crash Report (v1.4):
Result:                          0x4A8 (2168-0002)

Process Info:
    Process Name:                hbloader
    Program ID:                  010000000000100d
```

This is an Edizon data abort. You can even see edizon in the backtrace.