Atmosphere-NX/Atmosphere #742: Crash after trying to install app on homebrew store (0x4A8 / 2168-0002)

PurplePanda2001, 2019-12-09:
## Bug Report
### What's the issue you encountered?
Attempting to download NXShell on fresh atmosphere install from the homebrew appstore results in a crash (0x4A8 / 2168-0002).
### How can the issue be reproduced?
Prepare sd card with latest Kosmos release (v14.2), replace kosmos atmosphere with latest version (v0.10.0), and add exefs/fs patches from Modded_Warfare's guide.</br>
Boot unpatched switch into RCM, inject fusee primary payload, load homebrew menu, open appstore, and try to install NXShell.
### Crash Report

https://gist.github.com/PurplePanda2001/5de74d1b243a6f2e4cd9efd779614ac0

### System Firmware Version

9.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusèe
- Official release or unofficial build: 
  - Official release version 0.10.0
- Do you have additional kips or sysmodules you're loading: Yes, exefs/fs patches from Modded_Warfare's homebrew guide
- Homebrew software installed: appstore, EdiZon, Goldleaf, KosmosToolbox, KosmosUpdater, ldnmitm_config, Lockpick, NXDumpTool

### Additional context?

N/A

shchmue, 2019-12-09:
This is a known issue with hb-appstore they're working on mitigating for applet mode. The workaround is to open hbmenu by Title override (hold R and launch a title) instead of applet mode.

SciresM, 2019-12-09:
"Homebrew App Store" is not supported.

closed 2019-12-09 by SciresM