Atmosphere-NX/Atmosphere #745: (Help) [ERROR] Fatal error: Failed to stat stage2 (atmosphere/fusee-secondary.bin)!

ziraw, 2019-12-09:


ziraw, 2019-12-09:
oh it glitched i wanted t owrite something but anyway:
so i forgot to put my sd card in my switch and injcected the payload from TegraRcmGUI without having my sd card in now its stuck there after i out my sd card in while it was in rcmMode...
pretty stupid right?
what can i do now to boot atmosphere or even boot my switch in deafult settings? pls help
thanks in advanve to you guys

ziraw, 2019-12-09:
aslo tegraRcm doesnt regognize my switch anymore (while in rcm)

ziraw, 2019-12-09:
It suddenly bootet........

ziraw, 2019-12-09:
yeah...
its stuck in rcm and its saying this found nothing that helped

optimism247, 2019-12-09:
I can boot with fusee-primary.bin, but I can't launch in with the Hetake "Launch" function.

lamecake, 2019-12-09:
Had the same problem, the SD Card was not formatted properly (with the partition set to Primary and as FAT32), that fixed it for me.

ziraw, 2019-12-09:
yeah i havbe save files from a year ago on there so i cant format it :/

optimism247, 2019-12-09:
Can you not copy those files over? Or are they not saved onto the Switch itself?

ziraw, 2019-12-09:
yeah probably..
can copy them over then install atmosphere and then copy them back?

ziraw, 2019-12-09:
I just want acustom background thats it btw 
no harder modding

SciresM, 2019-12-09:
This is a user error. Seek help in the reswitched discord

closed 2019-12-09 by SciresM

ziraw, 2019-12-09:
wt*? what?

SciresM, 2019-12-09:
GitHub issues are for fixing atmosphere bugs, not for providing support.

ziraw, 2019-12-11:
so where am i supposed to find support?

el-emmo, 2019-12-11:
The ReSwitched Discord would be a place to ask (as Scires mentioned above).