Atmosphere-NX/Atmosphere #748: hbmenu freeze on 0.10

reaps101, 2019-12-10:
im running atmosphere 0.10.0 on firmware 4.0.1 and when in hbmenu and i push the home button my switch will lockup and i have to hold the power button for 12 seconds to turn it off 
there is no crash report

SciresM, 2019-12-10:
Does this only happen when opening hbl?

Does a hang occur opening album/games?

Do you have any custom system modules?

reaps101, 2019-12-10:
so in hbl i can press back and go through the folders soon as i press the home button it freezes
in the standard album if i press back it freezes
also in the controllers menu if i press close it freezes
i have no custom system modules only thing on sd card is atmosphere 0.10.0

SciresM, 2019-12-10:
Does the same hang occur when closing a game?

reaps101, 2019-12-10:
i didnt have any games installed but i just tried to put my zelda cartridge in and it wanted to download a game update and when i went to cancel it it froze

SciresM, 2019-12-10:
Please try to get a game launched -- I'm trying to tell if this is an applet exit problem or a general problem.

SciresM, 2019-12-10:
If you could message me on discord (SciresM#0524) that would be great.

reaps101, 2019-12-10:
sent u a message on discord think i did it right

alimao10, 2019-12-10:
I reported this problem but could not explain

alimao10, 2019-12-12:
https://www.youtube.com/watch?v=u6MbNYX1EfY

alimao10, 2019-12-12:
https://www.youtube.com/watch?v=AwF-F3qReX8

alimao10, 2019-12-12:
https://www.youtube.com/watch?v=Uq4ZDE391c8

alimao10, 2019-12-12:
https://www.youtube.com/watch?v=DSO__abNztM

alimao10, 2019-12-12:
sysnand 4.1atmosphere 0.9.4 not freezing

SciresM, 2019-12-12:
Please do not spam the issue tracker. The issue is known.

alimao10, 2019-12-12:
not spam sorry I'm showing the problem in videos, I will not report errors did not like

misson20000, 2019-12-13:
I can reproduce this on 3.0.0 by opening controller applet and attempting to exit it. I think it's a general applet exit problem.
Not sure whether this fall-through is intended or not, but adding a break there doesn't fix it.
https://github.com/Atmosphere-NX/Atmosphere/blob/f9a199557cf3b8269342ef0aa7b7e5f289bcad71/stratosphere/pm/source/impl/pm_process_manager.cpp#L436-L440

misson20000, 2019-12-13:
Suspecting that this codepath never actually cleans up process on <5.0.0, so the thread busyloops servicing the same process exit signal over and over. Not sure what the intended behavior here is though...
https://github.com/Atmosphere-NX/Atmosphere/blob/f9a199557cf3b8269342ef0aa7b7e5f289bcad71/stratosphere/pm/source/impl/pm_process_manager.cpp#L414-L438

commit 0a18c63f (2019-12-13): pm: fix process exit on < 5.0.0 (closes #748) (thanks @misson20000)
closed 2019-12-13 by SciresM