Atmosphere-NX/Atmosphere #749: Games installed on system software “Corrupted data”

EdwardElric64, 2019-12-10:
## Bug Report

### What's the issue you encountered?

After the headache of finding each and every individual file for Emuiibo on my sd, along with deleting my custom theme, I was finally able to successfully boot into CFW. But upon booting I have found that any program that I have installed to system memory (Goldleaf and Retroarch), the app crashes with the message “Unable to start software” and when trying to load it again it says “Corrupted data has been found” this goes for any homebrew app that I have installed to system memory

### How can the issue be reproduced?

Install nsp forward for any homebrew app (Goldleaf, Retroarch, etc) to system memory on 9.1.0 and atmosphere 0.10.0.

### Crash Report

None

### System Firmware Version

9.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusèe
- Official release or unofficial build: Official 0.10.0
- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: [ Goldleaf, Retroarch, freebird are ones that I use frequently]

### Additional context?

None

fincs, 2019-12-10:
> Homebrew software installed: [ Goldleaf,

Piracy is not supported.

closed 2019-12-10 by fincs

EdwardElric64, 2019-12-10:
I use the browser but ok...

James02m, 2019-12-19:
install the latest atmosphere sigpatches and you'll be fine, i had the same issue.

https://github.com/Joonie86/hekate/releases/tag/5.0.0J

fincs, 2019-12-19:
Homebrew is not installed. It is run using hbloader.
Sigpatches are 100% unnecessary for homebrew. They're only relevant for piracy. Link to piracy distribution of hekate has been removed.