Atmosphere-NX/Atmosphere #750: Homebrew installed to system memory “Corrupt”

EdwardElric64, 2019-12-10:
## Bug Report

### What's the issue you encountered?

After the headache of finding each and every individual file for Emuiibo on my sd, along with deleting my custom theme, I was finally able to successfully boot into CFW. But upon booting I have found that any program that I have installed to system memory (Goldleaf and Retroarch), the app crashes with the message “Unable to start software” and when trying to load it again it says “Corrupted data has been found” this goes for any homebrew app that I have installed to system memory

### How can the issue be reproduced?

Install nsp forward for any homebrew app (Goldleaf, Retroarch, etc) to system memory on 9.1.0 and atmosphere 0.10.0.

### Crash Report

None

### System Firmware Version

9.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusèe
- Official release or unofficial build: Official 0.10.0
- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: [ Goldleaf, Retroarch, freebird are ones that I use frequently]

### Additional context?

I do not use Goldleaf for any piracy whatsoever. I have used it to install homebrew nsp (such as Retroarch) along with the browser and nothing else. This has nothing to do with piracy and I wish to have some assistance with my issue please

fincs, 2019-12-10:
Just because you posted this twice doesn't mean that we are going to support piracy.

Piracy is not supported.

closed 2019-12-10 by fincs

EdwardElric64, 2019-12-10:
I literally just said it’s not used for piracy but thanks anyways I guess

fincs, 2019-12-10:
Installing any kind of NSP is not supported. Homebrew is only supported as NRO loaded through hbloader without involving "installable" NSPs.

EdwardElric64, 2019-12-10:
And that’s the answer I’m looking for. Thank you.