Atmosphere-NX/Atmosphere #751: cannot boot to atmosphere "a fatal error occurred"

juzzash, 2019-12-11:
## Bug Report

cannot boot to Atmosphere

### What's the issue you encountered?

i have installed the atmosphere 0.10.0  to my SD card (after formatting to FATex). (mac wont let me format to FAT32) when pushing the fusee-primary.bin payload via https://fusee-gelee.firebaseapp.com/# (i have a mac so this is al i could find that works) - My switch boots up with the atmosphere logo however then i get error message:

A fatal error ocurred when running atmosphere.
Title ID: 010041544d530000
Error Desc: std: :abort() called (0xffe)
Report saved to /Atmosphere/fatal_errors/report/00000000005a201c.bin

Press POWER to reboot

### How can the issue be reproduced?

ive tried a few times, even starting from scratch with the files on the sd card and it happens each time

### System Firmware Version
my switch is: 9.1.0

closed 2019-12-11 by juzzash

gretel, 2019-12-22:
@juzzash did you fix it?

Jenrikku, 2020-02-20:
I have the same problem, but I'm not using Mac. I used the Archive Bit Tool from Hekate and it stills giving me a boot error.

Using emuMMC 9.1.0 and sysMMC 9.1.0