Atmosphere-NX/Atmosphere #752: Atmosphere Fatal Error on Boot

aarongeorge, 2019-12-11:
## Bug Report

### What's the issue you encountered?
Atmosphere throws the following error when trying to boot from Hekate and Fusée:

```
A fatal error occurred when running Atmosphére.
Title ID: 010041544d530000
Error Desc: std::abort() called (0xffe)
```

[ Did you make any changes related to Atmosphère itself? ]</br>
No

### How can the issue be reproduced?
Boot to RCM
Launch Hekate
Boot Atmosphere through either fusèe-primary or emunand

### Crash Report
Not a crash report but a Fatal Error: https://gist.github.com/aarongeorge/a255f4f380b595477f12624d28b2b109</br>

### System Firmware Version

9.1.0</br>

### Environment?
Bootloader: Hekate
Release: 0.10.1 (Official)
Additional Kips/Sysmodules: N/A
Homebrew Installed: N/A

### Additional context?
This is happening after upgrading from 9.0.0 to 9.1.0 and then updating Atmosphere and Hekate to their latest versions.
I've had a hacked switch for a while now and this is the first time that upgrading has caused an issue.

SciresM, 2019-12-11:
The report you uploaded was corrupted when you uploaded it to gist. Please upload the raw binary, so that github doesn't UTF-8 encode it...

SciresM, 2019-12-11:
Staring at the corrupted binary, I think I know the issue anyway.

OpenFile is failing when trying to create the prodinfo backup (this should never fail, so we assert it succeeds).

I think your SD card has weird archive bits that prevent creation of files in the automatic backups folder.

Are you using macOS? Try using hekate's archive bit fixer, maybe. Also try using FAT32 instead of exFAT.

aarongeorge, 2019-12-11:
Yeah sorry, couldn't find a way to upload a file to gist, only copy and paste the contents.

I am on macOS, I'll try using the bit fixer and get back to you.

Thanks @SciresM

aarongeorge, 2019-12-11:
@SciresM I can confirm that running `sudo chflags -R arch` on all directories aside from the official Nintendo directory has solved the issue.

Thank you for your time and knowledge.

10/10 Support

SciresM, 2019-12-11:
No problem, glad the issue's resolved :)

closed 2019-12-11 by SciresM

jhawNS, 2019-12-21:
I got the same problem,
I tried using hekate archive bit fix and fromat FAT32,
didn't solve it.
how can i do?
[report_0000000005768c98.zip](https://github.com/Atmosphere-NX/Atmosphere/files/3990808/report_0000000005768c98.zip)

ObscenityIB, 2019-12-22:
Ive had this issue just now after having atmosphere crash during a large title install with Awoo, because Tinfoil is my usual go to, and the game crashes, so I tried another, I  fixed the archive bits but I'm still having the issue, been using 9.1.0 for a while now and this is the only time it has had an issue, just trying to stop a specific game from crashing after the ninty logo, no luck so far, other games i have are fine, including carts.

fincs, 2019-12-22:
Piracy is not supported.