Atmosphere-NX/Atmosphere #754: Games don't load AT ALL in emuMMC 0.10.1.

IAmArique, 2019-12-11:
I'm running Atmosphere 0.10.1 on 9.1.0. When I try to open a physical or (legally purchased) digital game while in emuMMC, the Switch locks up and I have to force shutdown the console, and doesn't give me a crash report since I just get a blank screen every time. The same thing happens if I try loading hbmenu while selecting a game.

fincs, 2019-12-11:
Do you have exFAT?

IAmArique, 2019-12-11:
Yes, I have exFAT. I'm using the RCMLoader dongle to load Atmosphere rather than hooking it up to my PC every time.

fincs, 2019-12-11:
Nintendo's exFAT driver is known to be bad & cause corruption issues. Please try with FAT32 as a troubleshooting step. Also, please answer the other questions (I edited my post, sorry if you didn't see it).

IAmArique, 2019-12-11:
Ok, I'll try changing my card to FAT32 and see if that works. I'll edit this comment once I test it.

As for the distribution, I downloaded Atmosphere from here on the Github page. I didn't get it from anywhere else.

fincs, 2019-12-11:
Alright. One more thing - file based emuMMC is known to be slower and less reliable. For the best chance of success, make sure you're using partition based emuMMC.

SciresM, 2019-12-12:
This was an issue with SaltySD/ReverseNX.

closed 2019-12-12 by SciresM