Atmosphere-NX/Atmosphere #755: Backlight only after injecting payload

DoloDomi, 2019-12-11:
one thing to point out i know the basic to a modded switch . so my switch was running atmosphere perfeclty fine i was on the latest version but i for once let my brother play it and idk what happen he gave me it with just the switch blacklight on , nothing was working so i hard reset since i can just inject the payload but once i did i just stays on the blacklight i had a backup of my files on my computer so i took my sd card reformat it to fat32 , put my files back and nothing still the same. ive tried other payload such as kosmos and hekete. for hekete itll load but nothing works after hitting launch i know i could load payloads so i use hekete then inside i tried loading fusee but back to backlight i go to. when trying kosmos i hit launch then cfw but it takes me back to hekete menu . i dont know whats wrong im running 9.1.0 on my switch if thats helps but yea i could provide videos or photos if needed i just really need help

SciresM, 2019-12-12:
1) use the issue templates

2) probably archive bit problems or fs corruption. That error happens when atmosphere can't  be loaded off the SD.

closed 2019-12-12 by SciresM

DoloDomi, 2019-12-12:
> * use the issue templates
> * probably archive bit problems or fs corruption. That error happens when atmosphere can't  be loaded off the SD.

I’m new on github I don’t know what you mean and it was perfectly fine when it was insert in the first place before everything happen . What would you recommend