Atmosphere-NX/Atmosphere #756: HBL time display is wrong

cucholix, 2019-12-11:
Using atmosphere 0.10.1, FW 9.1.0.
The time displayed in the upper right corner in HBL menu is off by 3 hours, example: if the time in system menu is 15:00, HBL will display 18:00. My timezone is Santiago (UTC -3:00).

Other people is experiencing the same issue https://gbatemp.net/threads/how-to-change-hbmenu-to-the-correct-time-zone-as-switch.553221/

My apologies if it’s not the right place to place this issue, not sure if HBL menu has its own github.

fincs, 2019-12-11:
HBL (hbloader) doesn't display any time. It's a module that is responsible for loading NRO. I think you refer to hbmenu, which is the graphical shell that sits on top of hbloader.

With that said - I cannot reproduce. My timezone is UTC+1. Make sure you aren't using any sort of dodgy third party recompiled or repackaged version of Atmosphère, nx-hbloader or nx-hbmenu.

cucholix, 2019-12-11:
I’m using clean atmosphere install, no dodgy forks. No matter what time zone I set, it always will display UTC +00:00

fincs, 2019-12-11:
Have you ever connected your Switch to the (real) Internet?

cucholix, 2019-12-11:
It’s always connected to internet (wifi)

yellows8, 2019-12-12:
With my normal TZ it's fine, however with my system's TZ changed to Santiago it displays UTC. https://github.com/switchbrew/libnx/issues

SciresM, 2019-12-12:
Yeah, I don't *think* this is an atmosphere issue so closing. Reopen in libnx?

closed 2019-12-12 by SciresM