Atmosphere-NX/Atmosphere #757: Hombrew menu not lauching

Burgos91, 2019-12-12:
After Updated to 10.1 even to the recently one everytime I try to launch hombre menu i give me this message:
Error:assetsInit() failed 0x80.
press the + button to exit.

fincs, 2019-12-12:
It looks like you have corruption of some sort. Please redownload and reextract Atmosphère direct from the official release. Also, if you have an exFAT formatted SD card, it might be related (exFAT is known to cause corruption issues), switch to FAT32 if you haven't already.

Burgos91, 2019-12-12:
I did now im making a backup to do a clean install and see if fix the problem and my SD is a 128gb ill have to do use a program to formated right?

fincs, 2019-12-12:
Yeah. I think Rufus can do the formatting.

Burgos91, 2019-12-12:
ok ill report as soon I finish and see if worked

Burgos91, 2019-12-12:
did everything you told me and it work for now thanx alot

closed 2019-12-12 by Burgos91