Atmosphere-NX/Atmosphere #758: Atmosphere hangs after the "A" logo.

baneonplane, 2019-12-12:
## Bug Report

### What's the issue you encountered?
Atmosphere does not boot into CFW. It hits the logo with a half-black half-white A, but not anything beyond that, including the Nintendo logo.

### How can the issue be reproduced?
Before this, I had a Switch running older Atmosphere. I updated using Choi to 9.1.0, then powered off, deleted all files in the atmosphere and sept folders, and replaced them with the files from the most recent .zip in releases. I can use fusee-primary to get into Atmosphere, but it hangs on a completely black screen after the aforementioned logo.

### Crash Report
No crash was logged.

### System Firmware Version
9.1.0

### Environment?
Using fusee-primary to boot. Official build of Atmosphere. I've removed most homebrews in /switch to minimize conflicts. I don't have emuNAND, only sysNAND, in case that matters.

### Additional context?
I've tried using both fusee-primary and the most recent version of Hekate to boot, but neither does. I've tried unsetting the archive bits using Hekate, but it doesn't seem like that helped. It's possible that I forgot to add something to my SD card that I need. I'll list the files in my Atmosphere and sept folders below to see if that could be the reason.
Atmosphere:
folders: config, config_templates, contents, fatal_errors, hbl_html, kip_patches (contains default_nogc)
files: fusee-mtc.bin, fusee-secondary.bin, hbl.nsp, reboot_payload.bin
sept (no folders):
files: payload, sept-primary, sept-secondary (all .bin), sept-secondary_00.enc, sept-secondary_01.enc

SciresM, 2019-12-12:
What's inside /atmosphere/contents?

May also be archive bit problems inside the atmosphere folder, but.

baneonplane, 2019-12-12:
These folders, no others, and no files. I'll try unsetting the archive bits again.
010000000000000D
0100000000000008
0100000000000032
0100000000000034
0100000000000036
0100000000000037

SciresM, 2019-12-12:
What SD card FS? If exFAT, try on FAT32?

baneonplane, 2019-12-12:
It's FAT32. Also for more info, each of those folders has a file called exefs.nsp. The ones ending in 32 and 37 have a folder called flags, each containing a file called boot2.flag.

SciresM, 2019-12-12:
Ah, FAT32, you mentioned.

Message me on discord (SciresM#0524). I can help you debug a lot less asynchronously than GitHub allows.

Thanks a ton for proper issue template usage.

SciresM, 2019-12-12:
SD card was exFAT, non-exFAT firm was installed.

closed 2019-12-12 by SciresM