Atmosphere-NX/Atmosphere #762: Build fails on Manjaro Linux

sogilbert, 2019-12-13:
Hi guys. When I try to build Atmosphere on Manjaro Linux, this error occurs:

arm-none-eabi-objcopy -S -O binary /home/xxxx/Atmosphere/sept/sept-secondary/sept-secondary.elf /home/xxxx/Atmosphere/sept/sept-secondary/sept-secondary.bin
built ... sept-secondary.bin
Traceback (most recent call last):
  File "/home/xxxx/Atmosphere/sept/sept-secondary/sept_sign.py", line 5, in <module>
    from Crypto.Hash import CMAC
ImportError: cannot import name 'CMAC' from 'Crypto.Hash' (/usr/lib/python3.8/site-packages/Crypto/Hash/__init__.py)
make[3]: *** [/home/xxxx/Atmosphere/sept/sept-secondary/Makefile:156: /home/xxxx/Atmosphere/sept/sept-secondary/sept-secondary.enc] Error 1
make[2]: *** [Makefile:129: build] Error 2
make[2]: Leaving directory '/home/xxxx/Atmosphere/sept/sept-secondary'
make[1]: *** [Makefile:8: sept-secondary] Error 2
make[1]: Leaving directory '/home/xxxx/Atmosphere/sept'
make: *** [Makefile:27: sept] Error 2

Could you help with this? Thank you.

HookedBehemoth, 2019-12-13:
You are missing pycryptodome for signing sept.
Since you most likely don't have the keys you shouldn't bother with that and rather export SEPT_00_ENC_PATH and SEPT_01_ENC_PATH to point to SciresM's signed sept.

SciresM, 2019-12-13:
Install pycryptodome.

Also, unless you have signing keys for sept, you'll want to build using the release sept binaries - else fusee you produce wont be usable on hardware.

SciresM, 2019-12-13:
Also, ninja'd.

closed 2019-12-13 by SciresM