Atmosphere-NX/Atmosphere #763: Issues with ams_mitm and system titles' RomFs replacement

XorTroll, 2019-12-13:
## Bug Report

### What's the issue you encountered?

Atmosphere's ams_mitm seems to abort when a custom system title reimplementation attempts to access a custom romfs.bin in /contents, even with the fsmitm.flag file.

### How can the issue be reproduced?

Create a custom library applet project (gist for my test: https://gist.github.com/XorTroll/2eec2791c64aa0fbbe87806706f7d47a), with its own romfs.bin too, and replace a system applet with the exefs.nsp and romfs.bin via LayeredFS and fs-mitm.

Placing the output exefs.nsp and romfs.bin (Makefile also builds a simple romfs.bin), with a fsmitm.flag replacing a library applet in /atmosphere/contents/<program-id> will make it abort when initializing RomFs.

This behavior worked perfectly on pre-0.9.x Atmosphere.

### Crash Report

This didn't generate

### System Firmware Version

9.0.0

### Environment?

- Launched by hekate 5.1.1, also tested with latest fusee-primary
- Tested on official Atmosphere, on both 0.10.0 and 0.10.1
- Only extra sysmodule I'm using emuiibo, but can confirm that isn't causing the issue, since I ran on the same issue without it too.
- Homebrew software installed: Some generic NROs, but didn't even launch any of them. My test replaced controller applet, so I just opened controller menu from HOME menu.

### Additional context?

- I reproduced this with two separate projects: uLaunch's main menu applet, which loads RomFs for its UI, and a special test made to test this. The moment `romfsInit()` (or a similar function I guess) is called, ams_mitm will abort.

SciresM, 2019-12-14:
I think this is correct behavior -- custom sysmodules do not have an NCA, and thus do not have romfs.

Thus openDataStorageByCurrentProcess "should" fail.

Depending on the title success but with base process romfs may also be okay, but I don't think that's desired from a design POV - it's a layeredfs artifact.

ams.mitm shouldn't fatal error though, only return an error code. You didn't upload the fatal error report, please do so.

SciresM, 2019-12-14:
"Sysmodule" --> meant sysapplet but it's the same thing here.

SciresM, 2019-12-14:
Also please note that fsmitm.flag is no longer a thing

commit 06e4b946 (2019-12-17): fs.mitm: only iter romfs dir if it exists (closes #763)
closed 2019-12-17 by SciresM