Atmosphere-NX/Atmosphere #765: Homebrew menu doesn't show up

Axodique, 2019-12-14:
### What's the issue you encountered?

I just updated Atmosphère (deleted atmosphere folder and put them all in), hekate and my console, i started my console and then i couldn't get into the homebrew menu.

### How can the issue be reproduced?
N/A

### Crash Report

N/A

### System Firmware Version

9.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: 
0.10.1
- Do you have additional kips or sysmodules you're loading:  none
- Homebrew software installed: hbl, goldleaf.

Ive also put a video to show what happens

Axodique, 2019-12-14:
https://youtu.be/R4aBnzdcWjo

fincs, 2019-12-14:
> goldleaf

> video containing a nsp version of hbmenu

Piracy is not supported.

closed 2019-12-14 by fincs