Atmosphere-NX/Atmosphere #766: HBL doesn't launch

darthpheonix, 2019-12-15:
## Bug Report

### What's the issue you encountered?

I set up a friend's switch with Atmosphere 0.10.1 and after a successful boot up whenever I try to go into HBL it crashes with a 2347-0003 error code.

### How can the issue be reproduced?

Not sure how it's being recreated, after booting the switch up from RCM and injecting the proper payload trying to go into HBL using the album and R it crashes showing the error message.

### Crash Report

https://gist.github.com/darthpheonix/a3846dc5f904e884166fb8b172d94586

### System Firmware Version

9.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
Included one in release section
- Do you have additional kips or sysmodules you're loading:
no
- Homebrew software installed:

### Additional context?

N/A

darthpheonix, 2019-12-15:
I also had the newest Tinfoil files included, after doing a clean install of all the files HBL launched like normal.

fincs, 2019-12-15:
https://github.com/switchbrew/nx-hbloader/blob/master/source/main.c#L307-L309

You're missing hbmenu.nro on the root of your SD card. If you're using a clean, official version of Atmosphère this shouldn't have happened, because Atmosphère bundles the latest versions of hbloader and hbmenu. Make sure you're using the official release; and also make sure you're using FAT32 (exFAT SD cards are known to be problematic).

And also:

> newest Tinfoil files included

Piracy is **not** supported.

closed 2019-12-15 by fincs