Atmosphere-NX/Atmosphere #767: Atmosphere's ams_mitm Error with specific modded game.

RoutineFree, 2019-12-16:
**What's the issue you encountered?**

Atmosphere's ams_mitm seems to abort in a specific modded game when game started

[Error message]
Panic occured while running atmosphere

TitleID: 010041544D530000
Error Desc: std:abort() (0xFFE)

**How can the issue be reproduced?**

Most of the modded game files, (except for the romfs.bin) are well applied but cause errors in certain games.

The following three games where the errors I found occur.
:
Resident Evil 6 - (01002A000CD48000)
Dragon's Dogma (JPN) - (010032C00AC58000) 
Football Manager 2020 Touch - (0100DA500EFB0000)

※ The following three games have an error, even if put an empty single folder in a romfs folder.
※ What's unusual is that the Dragon's Dogma (JPN) makes errors, but the Dragon's Dogma (USA) works well without errors.

**Crash Report**
N/A

**System Firmware Version**
9.1.0

**Environment?**
What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
Official release or unofficial build: Official release version - Atmosphère 0.10.1 
Do you have additional kips or sysmodules you're loading: sys-clk
Homebrew software installed: N/A

**Additional context?**
Additional info about your environment: N/A
[ Any other information relevant to your issue. ]

SciresM, 2019-12-17:
@RoutineFree Please upload the fatal error binaries that were generated by the reports.

RoutineFree, 2019-12-17:
fatal error binaries
:
https://github.com/RoutineFree/Test/blob/master/report_023928e43.bin
https://github.com/RoutineFree/Test/blob/master/report_df6b9fe060.bin

SciresM, 2019-12-17:
This crash occurs because parsing the romfs for those games causes fs.mitm to run into memory exhaustion.

Fixing this will require changing the way paths are stored in romfs building to be more memory efficient -- this will almost certainly lead to a minor performance hit...

SciresM, 2019-12-30:
@RoutineFree This is theoretically fixed in master with latest commits. Can you contact me on discord to test?

Joonie86, 2019-12-30:
I can confirm the latest commit fixed my issue with RE6 modding, Thanks @SciresM !

SciresM, 2019-12-30:
Yeah, RoutineFree also verified that football touch game is working now. What a pain...

closed 2019-12-30 by SciresM

Lioniox, 2021-11-25:
@SciresM I also have this problem but with Zelda BOTW can u help me and contact me on Discord too? My Discord is Lioniox#0454