Atmosphere-NX/Atmosphere #768: No Controller Input in Homebrew

Sevyn13, 2019-12-17:
I know homebrew apps need to be recompiled with newest libnx to fix this. But the release notes say there is a temporary hid-mitm to fix this but it's apparently not working for me. Is it a separate download I need to get or do I need to toggle it on? Thanks!

deadair3210, 2019-12-17:
What Homebrew are you using

SciresM, 2019-12-18:
Cannot reproduce -- make sure you're launching homebrew via latest hbl/hbmenu (these are included with atmosphere releases).

closed 2019-12-18 by SciresM

ELY3M, 2019-12-18:
What sysmodules are you running?
I have to get rid of old sysmodule that caused this same issue.  I
recompiled sysmodule again and it helps.

ericdude4, 2020-01-18:
@Sevyn13 Did you ever figure this out? Many of my homebrew apps don't recognize any controller inputs. Specifically retroarch (v1.7.7)

fincs, 2020-01-18:
On 9.x+ homebrew needs to be recompiled with latest libnx:

> Explicitly announce support for System/SystemExt layouts, which in turn fixes input on 9.0.0+. It is of utmost importance that all homebrew be recompiled as soon as possible in order to work properly on 9.0.0+.

ericdude4, 2020-01-18:
@fincs Hey, thanks for that tip. Can you send me a quick link on where I can read more about that?

fincs, 2020-01-18:
That's just an excerpt from the [libnx changelog](https://switchbrew.github.io/libnx/md__project_repo_checkout_Changelog.html). In particular that's in the changelog for the version prior to the latest at the time of writing; which was released in September of last year. There's been a major release since then (in December). I'm surprised they seem not to have recompiled their software yet.

uaevuon, 2020-01-18:
> @Sevyn13 Did you ever figure this out? Many of my homebrew apps don't recognize any controller inputs. Specifically retroarch (v1.7.7)

What...? Why do you use that old version and complain about waiting?
v1.8.4 is the latest and it is built with libnx 3.0

ericdude4, 2020-01-18:
@uaevuon Thanks for the tip.... I have been using the packages that get put together in this site https://sdsetup.com/console?switch. I guess I assumed they were all latest releases... I feel stupid... Thanks!