Atmosphere-NX/Atmosphere #771: Crash issue since change to contents folder.

LordBabbit, 2019-12-18:
## Bug Report

[ * ]</br>
[ Note: If the bug or crash you encountered is related to monster hunter generations ultimate; ]</br>
[ - software used to make "backups", edizon]</br>
[ - software explicitly distributed for piracy, etc n/a]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?
Ever since the change from the titles folder to the contents folder, whenever the game monster hunter generations ultimate is launched it crashed. After I deleted the cheats file from the contents folder the game would launch normally. It is a minor issue.
[ Describe the issue in detail and what you were doing beforehand. Cheating!]</br>
[ Did you make any changes related to Atmosphère itself? No]</br>
[ If so, make sure to include details relating to what exactly you changed. N/a]</br>

### How can the issue be reproduced?
Simply try to add a cheat text for mhgu in the new contents folder
[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

9.1.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee
- Official release or unofficial build: 
  - [ Official release version 0.10.1 (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. Kosmos preset from sdsetup, manually updated atmosphere files into it] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

LordBabbit, 2019-12-18:
Well... I finally learned that button mashing plays off. Force holding L caused the game to launch just fine suddenly.

closed 2019-12-18 by LordBabbit

shchmue, 2019-12-18:
all that does is disable cheats