Atmosphere-NX/Atmosphere #773: Bug Report

jhawNS, 2019-12-21:
What's the issue you encountered?
Atmosphere throws the following error when trying to boot from Hekate and Fusée:

A fatal error occurred when running Atmosphére.
Title ID: 010041544d530000
Error Desc: std::abort() called (0xffe)
[ Did you make any changes related to Atmosphère itself? ]

No

How can the issue be reproduced?
Boot to RCM
Launch Hekate
Boot Atmosphere through either fusèe-primary or emunand

Crash Report
Not a crash report but a Fatal Error: 
[https://github.com/jhawNS/66/blob/master/report_0000000005768c98.bin](url)
System Firmware Version
9.1.0

Environment?
Bootloader: Hekate
Release: 0.10.1 (Official)
Additional Kips/Sysmodules: N/A
Homebrew Installed: N/A

Additional context?
This is happening after upgrading from 9.0.0 to 9.1.0 and then updating Atmosphere and Hekate to their latest versions.
I've had a hacked switch for a while now and this is the first time that upgrading has caused an issue.
I tried using hekate archive bit fix 、 fromat FAT32、changed memory card
didn't solve it.

fincs, 2019-12-21:
ams_mitm is throwing an abort. Dupe of #767?

jhawNS, 2019-12-21:
> ams_mitm is throwing an abort. Dupe of #767?
Different
I stopped while loading the atmosphere
not running games

webben-de, 2019-12-21:
I run into the same issue. 
Is there any workaround?

ObscenityIB, 2019-12-23:
Having the same issue here, cant figure out how to fix it, tried the archive bit thing, no luck.
Really don't want to have to resort to empty nand backups.

OddieBaSs, 2019-12-23:
Same problem here, AMS is not running after updating to 9.1.0. Tried with Hetake 5.1.1, just AMS 0.10.1 files, etc.. same result: blank screen : (

ObscenityIB, 2019-12-24:
@webben-de @OddieBaSs 
My solution was clearing out `sdmc:/atmosphere/contents/` and replacing ONLY the stock contents and nothing additional.
I don't have any custom themes this time so I'm not sure what the source was.

Chen-Woo, 2019-12-24:
Evenings Friens Woo here to confirm the latest version has many bugs. Hold tight whilst the Elves CODE a solution and we need to be very cautious at these cold times. Frustration will show its self and brickings are possible. Woo

Chen-Woo, 2019-12-24:
[image]

deadair3210, 2019-12-24:
What the......?

jhawNS, 2019-12-24:
> @webben-de @OddieBaSs
> My solution was clearing out `sdmc:/atmosphere/contents/` and replacing ONLY the stock contents and nothing additional.
> I don't have any custom themes this time so I'm not sure what the source was.

I used this solution but it didn't work. :(

CTCaer, 2019-12-24:
The trick is to **run both** of the following options. Not just one or the other.
[image]

ObscenityIB, 2019-12-25:
> The trick is to **run both** of the following options. Not just one or the other.
tried that, did absolutely nothing for me

CTCaer, 2019-12-26:
If this doesn't fix it, it means that something in the now called contents folder is not compatible.
So step 2 is to delete that folder and put the one from the release manually. But you will probably need to run the archive bit options again after that. (Most OSes add the archive bits when a folder is created).

jhawNS, 2019-12-26:
> The trick is to **run both** of the following options. Not just one or the other.
I used this solution and replaced this folder "sdmc:/atmosphere/contents/" 
but ...
[image]

Is there other workaround?

SciresM, 2019-12-30:
This console has a corrupted PRODINFO. Somehow the string ```conUrl": null,
        ``` is being used as a serial number, which is obviously invalid. This causes an abort early in boot.

Un-corrupt your PRODINFO...?

SciresM, 2019-12-30:
Feel free to talk to me on discord about wtf is going on with your console.

closed 2019-12-30 by SciresM

jhawNS, 2019-12-30:
> Feel free to talk to me on discord about wtf is going on with your console.
Thank you,
What is your discord ID?

SciresM, 2019-12-30:
SciresM#0524

silentbomber25, 2019-12-30:
@SciresM I am having the same error and I am trying to reach out to you on discord.

WhyNintendo, 2020-02-29:
> SciresM#0524

Hey Scires I am having a similar problem i sent you a request on discord if you are able to reach out

g2nd, 2020-05-04:
Hi SciresM#0524, also having the same problem "A fatal error occurred when running Atmosphère. Title ID : 010041544D530000" after the daughter updated the firmware to 10.0.2.

g2nd, 2020-05-04:
> 
> 
> Hi SciresM#0524, also having the same problem
> 
> A fatal error occurred when running Atmosphère.
> Title ID : 010041544D530000"
> Error Desc : std::abort() called (0xffe)
> 
> Can you hint which things to check to fix this?
> i'm also unable to send you a DM in Discord via SciresM#0524.

Fixed it.  Reason: updating firmware online changes the filesystem to FAT32.  Had to use a FAT32 SD with Atmosphere.   Once loaded, used ChoiDujourNX to downgrade back to 10.0.1 in Exfat format.

g2nd, 2020-05-04:
ISSUE: Updated NSW to 10.0.2 via the internet

ERROR
A fatal error occurred when running Atmosphère.
Title ID : 010041544D530000"
Error Desc : std::abort() called (0xffe)

REASON: Updating Firmware online changes file system from Exfat to Fat32

SOLUTION:
1. Use and setup a fat32 SD card containing atmosphere.
2. Downgrade firmware to 10.0.1 using ChoiDujourNX using Exfat format.  (I encountered error updating it to 10.0.2 Exfat so I had to downgrade).
3. Put back your old Exfat SD card that was working previous to the update.
4. Update firmware to 10.0.2 using ChoiDujourNX using Exfat format.
5. Done.

deadair3210, 2020-05-04:
Likely not an error with atmosphere if you used the  online update. For some reason the switch didn't install the exfat fs driver when it updated, which made it fail to boot afterwards. Either install the exfat driver or format to fat32, which is recommended anyway as N's driver is hot garbage

thierry77c, 2020-05-04:
PROBLÈME: mise à jour de NSW vers 10.0.2 via choix du jour

ERREUR
Une erreur fatale s'est produite lors de l'exécution d'Atmosphère.
ID de titre: 010041544D530000 "
Erreur Desc: std :: abort () appelé (0xffe)

hack en sysnand (13 efuses)

Gnzloth, 2020-05-09:
Any solution?? I have the same fatal error, any channel in discord to fix this? Im Gnzloth#0220

Manuel-Flores-F, 2020-05-14:
This is usually due to an archive metadata bit used by some OS’s that Horizon simply can’t deal with. Here’s how to fix it:

Get your switch in RCM mode.
Inject Hekate payload
Tap on "Tools" on top of the hekate screen.
Tap on "Archive Bit - AutoRCM" on the right bottom.
Tap on "Unset archive bit" button. This might take a while, be patient.

MangoScango, 2020-05-14:
I'm having this issue on Atmosphere 12, HOS 10.0.0

Did the archive bit fix with Hekate, didn't fix it. Tried an older version of Hekate in case there was some sort of issue with the new "all in one" bit fixer. Still no luck. I'm currently backing up my SD card, and will be formatting to FAT32 (from exFat) to see if that does anything.

Real stumped on this one. I don't recall changing anything, just found my switch in RCM one day and couldn't boot it. Here's the error dump if it helps.

[report_0000000005ff1b1f.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4631444/report_0000000005ff1b1f.zip)

jonesmith518, 2020-06-27:
> The trick is to **run both** of the following options. Not just one or the other.
> [image]

u just saved my ass