Atmosphere-NX/Atmosphere #774: crash on load up

pito4evr, 2019-12-21:
i just updated the switch to 9.1 and am running 10.1 but it crashes after the nintendo logo
[image]

fincs, 2019-12-21:
qlaunch crashed. Remove custom themes and try again.

pito4evr, 2019-12-21:
how would i remove the custom themes since i cant load it up

fincs, 2019-12-21:
Take out the SD card, pop it in your pc and delete /atmosphere/contents/0100000000001000.

pito4evr, 2019-12-21:
that worked. thanks. i also just got a message saying game cart could not be read.

fincs, 2019-12-21:
Edit /atmosphere/config/BCT.ini and add nogc=0 to the end of it.

pito4evr, 2019-12-21:
i dont really care about the fuses since i dont plan on downgrading. Also would i reinstal themes like before or would have have to do an extra step on the newfirmware

fincs, 2019-12-21:
You'd need to get new themes that are built for your system version. For this reason, custom themes aren't really recommended; they're more of a hassle than they are worth.

Also, since this issue is now solved I'm closing it. This issue tracker is for reporting confirmed Atmosphère bugs and feature requests. Next time an issue arises, please check first the provided user support link that appears when you try to create an issue ("Please ask questions here...").

closed 2019-12-21 by fincs