Atmosphere-NX/Atmosphere #775: Complete system lockup in Mario Odyssey

j-james, 2019-12-23:
## Bug Report

### What's the issue you encountered?

When playing Mario Odyssey, the game froze at the main screen, with the Switch unresponsive to any controls. I eventually had to force-shutdown the console by holding down power for ~15 seconds.

No changes have been made to Atmosphere itself.

### How can the issue be reproduced?

I haven't been able to reproduce this issue. 

### Crash Report

Atmosphere didn't go to a crash screen - the entire Switch just was unresponsive. Both of my crash reports in `atmosphere/crash_reports` came from an unrelated incident (me trying to launch Kosmos Toolbox as a standalone).

### System Firmware Version

9.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by?
  - fusee-primary initially, although I may have rebooted to Hekate at some point. Both were updated for Atmosphere 0.10.0.
- Official release or unofficial build
  - Official release version 0.10.1 from GitHub.
- Additional kips or sysmodules?
  - sys-con
  - sys-ftpd-light
- Homebrew software installed?
  - Checkpoint
  - Goldleaf (no sigpatches)
  - MelonDS
  - RetroArch

### Additional Context?

The symptoms of this issue seem similar to #748 and #322.

I'm unfortunately not going to be very helpful in debugging, as I don't have a way to relaunch Atmosphere right now, but I can pull files from the SD card.

fincs, 2019-12-23:
Remove custom sysmodules, and try again.

j-james, 2019-12-23:
SMO is physical, SD card is FAT32. 
Unfortunately, I don't have a payload injector with me, and I wasn't able to boot back to Atmosphere.

ELY3M, 2019-12-25:
have you added any mods for the super mario odyssey?

j-james, 2019-12-25:
No modifications have been made to the system or games.

SciresM, 2019-12-30:
```
Additional kips or sysmodules?
sys-con
sys-ftpd-light
```

Does issue occur without additional sysmodules?

j-james, 2019-12-30:
I'm unable to test. Unfortunately, I don't have a payload injector with me, and I wasn't able to boot back to Atmosphere.

That might be a valid enough reason to close this.

SciresM, 2019-12-30:
Unable to boot back to ams kind of suggests there's weirder stuff going on, yeah. If you get it reproducing make a new issue?

closed 2019-12-30 by SciresM

j-james, 2019-12-31:
Unable to boot back to AMS just means I don't have my payload injector : /

If I can reproduce, will do.

teamjonastan, 2020-01-19:
I had the same exact issue. It happens when I accessed the Action Guide. The issue is with the sys-modules; sys-ftpd-light and sys-con. If you remove them, the game shouldn't freeze up anymore.

joshuajaharwood, 2020-04-20:
> 
> 
> I had the same exact issue. It happens when I accessed the Action Guide. The issue is with the sys-modules; sys-ftpd-light and sys-con. If you remove them, the game shouldn't freeze up anymore.

Fixed for me just by disabling sys-con. Strange bug!